react-native-camera/react-native-camera #1335: How to save image

burhanahmed92, 2018-03-09:
How to save captured image in gallery with RNCamera api. I was previously using Camera api and i was saving images like this 
```
this.camera.capture()

            .then((data) => console.log(data))

            .catch(err => console.error(err));
```

But new RNCamera api is doing like this
```

 if (this.camera) {
            const options = { quality: 0.5, base64: true };
            const data = await this.camera.takePictureAsync(options)
            console.log(data.uri);
        }
```

What is do with data.uri ? How i can i save it in a given path. Kindly guide

jgfidelis, 2018-03-09:
Use (react-native-fs)[https://github.com/itinance/react-native-fs].

Use the camera roll api and pass the data.uri to its method.

There are many examples of this on the internet.

jgfidelis, 2018-03-09:
You may also use RN's own `saveToCameraRoll` method:
https://facebook.github.io/react-native/docs/cameraroll.html#savetocameraroll

closed 2018-03-09 by jgfidelis