react-native-camera/react-native-camera #1337: react-native-camera record video crashes while capturing video

RCTnativeapp, 2018-03-09:
Hi, I am using in reactnative for recording video. It works in iOS and but not in Android
"react-native-camera": "^0.12.0",

When I start recoding video, from camera while capturing the timer, app crashes(attached crash log).
Please help.
[crashlog.txt](https://github.com/react-native-community/react-native-camera/files/1798368/crashlog.txt)

Thanks,
Sipra

jgfidelis, 2018-03-10:
try to update your package to 1.0.1 and use RNCamera

RCTnativeapp, 2018-03-10:
Thanks. will update the package.

Any clue on below error while recording video:
 startCopy UserHandle{0}: MeasureParams{e370b50 com.android.soundrecorder}
03-09 16:22:13.918 7275-7275/? E/AEE/DUMPSTATE: copy_file: Copy /sys/kernel/debug/page_recorder/Usage_rank to SYS_PAGE_RECORDER failed(2), No such file or directory

fazeelzama20, 2018-03-16:
When I start video recording then this error occurred.
Error: start failed.
    at createErrorFromErrorData (index.bundle:1807)
    at index.bundle:1760
    at MessageQueue.__invokeCallback (index.bundle:2118)
    at index.bundle:1935
    at MessageQueue.__guard (index.bundle:2053)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (index.bundle:1934)
    at debuggerWorker.js:72

ChristineTzeng, 2018-04-06:
Does that mean the video recording feature for RCTCamera doesn't work anymore?

coryarmbrecht, 2018-04-07:
I'm hitting the same issue, but I don't get any error output.

SimonErm, 2018-10-11:
https://github.com/react-native-community/react-native-camera/pull/1835

closed 2018-10-11 by SimonErm