react-native-camera/react-native-camera #1340: Crash on take picture

emretekince, 2018-03-12:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

RNCamera

### Steps to reproduce
1. open camera
2. takePictureAsync()
3. crash

### Expected behaviour
Tell us what should happen

### Actual behaviour
app crashes when calling takePictureAsync function

### Environment
- **Node.js version**:
- **React Native version**: 0.54.1
- **React Native platform + platform version**: Android 4-8

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

brunolemos, 2018-03-12:
I had a `java.lang.NoClassDefFoundError: Failed resolution of: Landroid/support/media/ExifInterface;` crash, this is fixed by adding this on `app/build.gradle`:

```diff
 compile (project(':react-native-camera')) {
         exclude group: "com.android.support"
         exclude group: "com.google.android.gms"
+        compile 'com.android.support:exifinterface:27.+'
     }
     compile ("com.google.android.gms:play-services-vision:11.8.0") {
         force = true;
     }
```

But after this I have this crash: `java.lang.NullPointerException: Attempt to invoke virtual method 'int android.graphics.Bitmap.getWidth()' on a null object reference`

Because for some reason `mBitmap = BitmapFactory.decodeByteArray(mImageData, 0, mImageData.length);` is returning null. I don't have a fix.

brunolemos, 2018-03-12:
@emretekince do you have this on simulator only or also on a real device?

brunolemos, 2018-03-12:
@emretekince @ratkus91 do you have this on Emulator only or also on a real device?

emretekince, 2018-03-12:
@brunolemos Both

gla5001, 2018-03-12:
i'm having same issue

Traviskn, 2018-03-13:
I'm having problems with crashing only when trying to use face detection features on Android.  This is the error message I get:

```
03-12 18:52:53.155 9726-9775/com.universal E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #1
...
java.lang.RuntimeException: An error occured while executing doInBackground()
...
Caused by: java.lang.NoClassDefFoundError: com.google.android.gms.internal.zzbjt
```

brunolemos, 2018-03-13:
Update: The crash I was facing only happens on emulator, worked fine on device.

jgfidelis, 2018-03-13:
@Traviskn probably related to your build/gradle. How is it? compile('reacti-native-camera') {...} ?

Traviskn, 2018-03-13:
Here are my dependencies in my gradle file:
```
dependencies {
    compile project(':react-native-device-info')
    compile project(':react-native-wheel-picker-android')
    compile project(':react-native-fetch-blob')
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
    }
    compile ("com.google.android.gms:play-services-vision:10.2.0") {
        force = true;
    }
    compile project(':react-native-vector-icons')
    compile project(':react-native-sensitive-info')
    compile project(':react-native-i18n')
    compile project(':react-native-threads')
    compile project(':react-native-sqlite-storage')
    compile fileTree(include: ['*.jar'], dir: 'libs')
    compile 'com.android.support:appcompat-v7:26.0.2'
    compile 'com.facebook.react:react-native:+'
}
```
I have quite a few dependencies already, after some googling I'm afraid I might be running into the Android 64K method limit, I may need to enable multi-dexing

sibelius, 2018-03-14:
Check RNCamera example to see if the deps are correct

closed 2018-03-14 by sibelius

emretekince, 2018-03-14:
@brunolemos it's fixed when I add **compile 'com.android.support:exifinterface:27.+'**. also working in emulator (Genymotion)

victorpiccoli, 2018-03-15:
@Traviskn I had a similar error, fixed by removing 

 {
    	exclude group: "com.google.android.gms"
	compile 'com.android.support:exifinterface:27.+'
	}
    compile ("com.google.android.gms:play-services-vision:10.2.0") {
    	force = true;
	}
in app module of build.gradle

Traviskn, 2018-03-21:
@victorpiccoli Thanks for the tip!

filipencus, 2018-05-15:
Still crashing without any information to debug. Both emulator and real device crash when take picture.
RNCamera: 1.1.2
React Native: 0.55