react-native-camera/react-native-camera #1342: App crashes when react-native-navigation is installed

arapocket, 2018-03-12:
### Steps to reproduce
1. Create empty project and manually install react-native-camera and verify you can install it correctly
2. Create empty project and manually install react-native-navigation and verify you can install it correctly
3. Try to add react-native-navigation to the react-native-camera project

### Expected behaviour
App should run with no problems

### Actual behaviour
App builds successfully but crashes before opening

### Environment
- **Node.js version**: v8.9.4
- **React Native version**: 0.54.0
- **React Native platform + platform version**: Android simulator (most recent version?)

### react-native-camera
**Version**: npm version
**____________________________________________________________________________________________________**
**Relevant Code**
**____________________________________________________________________________________________________**

**MainApplication.java**

```
package com.androidtest;

import android.app.Application;

import com.facebook.react.ReactApplication;
import com.facebook.react.ReactNativeHost;
import com.facebook.react.ReactPackage;
import com.facebook.react.shell.MainReactPackage;
import com.facebook.soloader.SoLoader;

import java.util.Arrays;
import java.util.List;
import org.reactnative.camera.RNCameraPackage;
import com.reactnativenavigation.NavigationApplication;

public class MainApplication extends NavigationApplication {

	@Override
	public boolean isDebug() {
		// Make sure you are using BuildConfig from your own application
		return BuildConfig.DEBUG;
	}

    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(),
          new RNCameraPackage()
      );
    }

    @Override
    public List<ReactPackage> createAdditionalReactPackages() {
      return getPackages();
    }

    @Override
public String getJSMainModuleName() {
	return "index";
}

}

```

**MainActivity.java**

```
import com.reactnativenavigation.controllers.SplashActivity;

public class MainActivity extends SplashActivity {

}
```

arapocket, 2018-03-13:
Well it miraculously started working so closing for now but if anyone has similar issues please chime in.

closed 2018-03-13 by arapocket

FlaviooLima, 2018-03-16:
Well, it's happening to me -.-
Any suggestions?

FlaviooLima, 2018-03-16:
Well For future [reference](https://github.com/wix/react-native-navigation/issues/2349), maybe something here can help out.

arapocket, 2018-03-16:
Your logcat in Android Studio will help you immensely. But all of these problems stem from unnecessary play service version trouble (I hate android so much now). Here's the important part of my app/build.gradle which is working now.

Heed my advice and stick to 11.4.0 for `details.useVersion('11.4.0')` as any other version was crashing react-native-maps.

```
configurations.all {
    resolutionStrategy.eachDependency { DependencyResolveDetails details ->
        if (details.getRequested().getGroup() == 'com.google.android.gms') {
            details.useVersion('11.4.0')
        }
            def requested = details.requested
        if (requested.group == 'com.android.support') {
            if (!requested.name.startsWith("multidex")) {
                details.useVersion '26.0.1'
            }
        }
    }
}

```