react-native-camera/react-native-camera #1343: Support aspect fill on RNCamera

brunolemos, 2018-03-13:
### Which implementation are you using

RNCamera

### Steps to reproduce
1. Create a square RNCamera component

### Expected behaviour
- The full square being filled without distorting camera image.
- Support `aspect` prop like RTCCamera does. (then I'd use `Camera.constants.Aspect.fill`).

### Actual behaviour
Image is distorted, the same way it would if we use `Camera.constants.Aspect.stretch` on RCTCamera.

### Environment
- **Node.js version**: 8.9.0
- **React Native version**: 0.54.0
- **React Native platform + platform version**: Android 7

### react-native-camera
**Version**: 1.0.2

chenkang2208, 2018-03-14:
I'm getting the same issue after updated from RTCCamera to RNCamera.

[image]

[image]

mjaffa-figure, 2018-03-27:
Yeah This would be nice to get fixed, as I am getting this issue too. Works great on iOS, and getting ready to pass on this component and either build our own or look for something else out there. But this is by far the best I have found out there.

jgfidelis, 2018-04-04:
please try on master branch

waitopiggu, 2018-04-09:
This is still a problem as of the master branch be3e0ebfb8ff42a48211b55054325548cd304694
[image]

waitopiggu, 2018-04-11:
@aalices That's one way of looking at it, but it's not a good solution as it leads to unnecessary complexity. It would be better to just get the Android implementation to work correctly.

The logic [here](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraView.java#L172) needs a little more work. I will have a look at it when I get the chance or when I start testing on Android proper.

nberktumer, 2018-04-11:
@jgfidelis I tried using RCTCamera and it was working fine with aspect prop.

nberktumer, 2018-04-11:
@waitopiggu I had a look at the part that you said and changed the content of the if statement there. It's working fine on me now. I didn't tested the else statement but if statement is working fine.

```
if (height / width < ratio) {
    correctHeight = (int) height;
    correctWidth = (int) (height / ratio);
} else {
    correctHeight = (int) (width * ratio);
    correctWidth = (int) width;
}
```

waitopiggu, 2018-04-12:
@nberktumer I'll try it out tomorrow and with a few different preview sizes to see what happens.

waitopiggu, 2018-04-14:
I spent a couple of hours this afternoon working on this and it's working pretty well.

If someone else wants to test it out on another phone before I make a PR, that would be great.

https://github.com/waitopiggu/react-native-camera

---

[image]

avshenuk, 2018-04-14:
Nice, thanks, man. I’ll test it out in a while!

sibelius, 2018-04-14:
it looks like a good approach, tks for the work

send a PR so more people could easily test it

can u also share this example app?

waitopiggu, 2018-04-14:
@sibelius Done and https://github.com/waitopiggu/rn-camera-android-cropping-test

closed 2018-04-14 by sibelius

brunolemos, 2018-04-14:
@sibelius can you make a new npm release?

pkyeck, 2018-09-20:
@waitopiggu your example isn't working for me - it's resizing ok, but the preview always has a 4:3 ratio. Is this because my device doesn't support 1:1? Is this even a valid ratio?

PR #1468 (2018-04-14): fix(Android): image stretched instead of cropped
To resolve #1343
- Used the logic from RCTCameraView, but that only worked correctly in Landscape mode.
- Added an orientation check and more logic for Portrait mode.