react-native-camera/react-native-camera #1344: Demo can't run!

AncientMing, 2018-03-13:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

n1ru4l, 2018-03-13:
I also noticed that. I will try to fix it this week

jgfidelis, 2018-03-13:
https://github.com/react-native-community/rncamera-example

Is the oficial RNCamera example app

n1ru4l, 2018-03-13:
@jgfidelis You should remove the Example folder in this repository and link the demo project

sibelius, 2018-03-14:
@n1ru4l feel free to send a pr removing this demo from this repo

jdeanwaite, 2018-03-23:
@sibelius I created PR #1396

closed 2018-04-15 by jgfidelis