react-native-camera/react-native-camera #1347: File not found

AncientMing, 2018-03-14:
#import <GoogleMobileVision/GoogleMobileVision.h>

File not found!!

### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-03-14:
Are you using face detection?

sibelius, 2018-03-14:
follow docs to integrate properly

closed 2018-03-14 by sibelius

jiaoxuebing2014, 2018-03-16:
[image]

#import <GoogleMobileVision/GoogleMobileVision.h>
File not found!!

pohsiu, 2018-03-20:
+1 same as @jiaoxuebing2014