react-native-camera/react-native-camera #1348: Samsung picture rotated when picture is taken

ejkkan, 2018-03-14:
### Which implementation are you using
*RNCamera*

### Steps to reproduce
1. take picture on a samsung phone

### Expected behaviour
When a picture is taken the picture should be faced the same way as the preview has taken it, the default behavior of any other android or ios phone. Tested on sony and honor phone, which works properly.

### Actual behaviour
When picture is taken on an samsung, the picture is rotated x amount of degrees rightward.

[image]

[image]

### Environment
- **React Native version**: 0.51.0
- **React Native platform + platform version**: Android version *any*, Only samsung phones. Tested on S8 and Galaxy J5

### react-native-camera
**Version**:  git+https://git@github.com/react-native-community/react-native-camera and 1.0.2 tested

jgfidelis, 2018-03-14:
can you test with the fixOrientation option?
like:
```
options = { fixOrientation: true };
const data = await this.camera.takePictureAsync(options);
...
```

closed 2018-03-14 by sibelius

eliaahadi, 2018-03-24:
@jgfidelis I tried that and it still rotates incorrectly. Any other suggestions?

zthornto, 2018-06-27:
Not sure why this has been marked as closed. Am experiencing the same issue on an S7 edge. Has anyone discovered a fix yet? Thanks.

byunghyunpark, 2018-07-20:
@ziks21 How did you solve the above issue?

fozzarelo, 2018-09-05:
Has this issue been addressed?

hielfx, 2019-03-11:
This issue is still happening today, has anybody found a solution yet? I'm experiencing this issue with a Samsung Galaxy S8 device and Xiaomi Mi A1

skizzo, 2019-03-19:
Anybody managed to fix this?

vinnyboySherb, 2019-03-19:
To skip pre-process from your device, you can do that :
`await this.camera.takePictureAsync({skipProcessing: true})`
That way the picture will be save has it will be taken by the user.
Find here : https://docs.expo.io/versions/latest/sdk/camera/