react-native-camera/react-native-camera #1350: Fatal Exception: java.lang.RuntimeException: No camera available.

sibelius, 2018-03-14:
```
Fatal Exception: java.lang.RuntimeException: No camera available.
       at com.google.android.cameraview.Camera2.chooseCameraIdByFacing(Camera2.java:584)
       at com.google.android.cameraview.Camera2.start(Camera2.java:272)
       at com.google.android.cameraview.CameraView.start(CameraView.java:245)
       at org.reactnative.camera.RNCameraView.onHostResume(RNCameraView.java:297)
```

d41z, 2019-04-16:
hi @sibelius did you find the solution to this?

sibelius, 2019-04-16:
camera2 is still buggy I guess

closed 2019-06-22 by stale[bot]

cristianoccazinsp, 2019-11-20:
Just got this error using Camera1:

```
java.lang.RuntimeException No camera available. 
    Camera1.java:862 com.google.android.cameraview.Camera1.chooseCamera
    Camera1.java:184 com.google.android.cameraview.Camera1.start
    CameraView.java:314 com.google.android.cameraview.CameraView.start
    RNCameraView.java:496 org.reactnative.camera.RNCameraView$4.run
    Handler.java:883 android.os.Handler.handleCallback
    Handler.java:100 android.os.Handler.dispatchMessage
    Looper.java:214 android.os.Looper.loop
    HandlerThread.java:67 android.os.HandlerThread.run
```

Looks like the device definitely failed to get any camera. This happened on a Pixel 3, which clearly has cameras. Not sure why this is happening, but the runtime exception might be a bit too much.

https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L860

Considering how unlikely this is to happen, probably not worth fixing, but I wonder what could cause it.

sibelius, 2019-11-20:
it works with camera2 on Pixel 3?

cristianoccazinsp, 2019-11-20:
Error is from Camera1, and just some random error. Usually works fine.