react-native-camera/react-native-camera #1352: Android build fails with 'missing return value' error

gla5001, 2018-03-15:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Build android
2.
3.

### Expected behaviour
The android build should be successful

### Actual behaviour
The android build fails

```
> Task :app:processDebugGoogleServices
Parsing json file: /Users/gregalexander/Dev/fooder/android/app/google-services.json

> Task :react-native-camera:compileDebugJavaWithJavac
/Users/gregalexander/Dev/fooder/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/MutableImage.java:222: error: incompatible types: missing return value
          return;
          ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
1 error

FAILURE: Build failed with an exception.
```

Maybe `convertExposureTimeToDoubleFormat` should return empty string when `!exposureTime.contains("/")` is true?

### Environment
- **Node.js version**: 8.9.1
- **React Native version**: 0.54.2
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**:  master

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

gla5001, 2018-03-15:
if the fix for this is to just return an empty string in that case, i can open up a PR. Just let me know

sibelius, 2018-03-15:
please do it

gla5001, 2018-03-15:
👍

ivansnek, 2018-03-15:
I'm facing the same error, is there a fix for this?

almasaeed2010, 2018-03-15:
Tested #1353 and it works 👍

sibelius, 2018-03-15:
fixed, tks

closed 2018-03-15 by sibelius