react-native-camera/react-native-camera #1357: 【RNCamera】Camera is being used after camera.release() was called

ZionChang, 2018-03-16:
On Android the Camera occasionally crashes with the following exception:

### Stacktrace

       Fatal Exception: java.lang.RuntimeException: Camera is being used after Camera.release() was called
       at android.hardware.Camera.setHasPreviewCallback(Camera.java)
       at android.hardware.Camera.access$600(Camera.java:164)
       at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1270)
       at android.os.Handler.dispatchMessage(Handler.java:111)
       at android.os.Looper.loop(Looper.java:224)
       at android.app.ActivityThread.main(ActivityThread.java:5929)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1113)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:879)

### Environment
- **React Native version:0.50.3
- **React Native platform + platform version**: Android 5.1

### react-native-camera
**Version**: 0.10.0

jgfidelis, 2018-03-16:
What react-camera-camera version are you using?

ZionChang, 2018-03-18:
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera"

Dallas62, 2018-04-18:
Same on:
```
react-native-camera@^1.0.2:
  version "1.0.2"
  resolved "https://registry.yarnpkg.com/react-native-camera/-/react-native-camera-1.0.2.tgz#490e75b14196af0ec972a759bb6027693ad4aad6"
  dependencies:
    lodash "^4.17.4"
    prop-types "^15.5.10"
```

Will release a new version this week with:

```
react-native-camera@^1.1.1:
  version "1.1.1"
  resolved "https://registry.yarnpkg.com/react-native-camera/-/react-native-camera-1.1.1.tgz#307e8239800e6282298bbb1807d72c026899c03f"
  dependencies:
    lodash "^4.17.4"
    prop-types "^15.5.10"
```

I will give you feedback if the issue is still there.

n1ru4l, 2018-09-24:
@Dallas62 Do you still experience this issue?

Dallas62, 2018-09-25:
@n1ru4l Hi! Can't tell you, I'm not working for the company which own the application anymore...

n1ru4l, 2018-09-25:
Please feel free to reopen/create a new issue if anyone ist still experiencing this issue.

closed 2018-09-25 by n1ru4l

subugwriter, 2018-10-31:
I am having same issue on V1.1.4. Android camera occasionally crashes with same exception:

                                                                            java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                                at android.hardware.Camera.setHasPreviewCallback(Native Method)
                                                                                at android.hardware.Camera.-wrap0(Unknown Source:0)
                                                                                at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1122)
                                                                                at android.os.Handler.dispatchMessage(Handler.java:105)
                                                                                at android.os.Looper.loop(Looper.java:164)
                                                                                at android.app.ActivityThread.main(ActivityThread.java:6938)
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
                                                                                at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)