react-native-camera/react-native-camera #1358: barcode/QR scaning not working

ketankulkarni, 2018-03-16:
```
import React, { Component } from "react";
import {
  StyleSheet,
  Dimensions,
  Text,
  View,
  Image,
  TouchableOpacity
} from "react-native";
import Button from "../components/Button";
import { RNCamera } from "react-native-camera";

class TakePicture extends Component {
  constructor(props) {
    super(props);
    this.state = {
      qrcode: ""
    };
  }

  takePicture = async () => {
    try {
      const data = await this.camera.takePictureAsync();
      console.log("Path to image: " + data.uri);
    } catch (err) {
      // console.log('err: ', err);
    }
  };

  onBarCodeRead = e => {
    this.setState({ qrcode: e.data });
    console.log(this.state.qrcode);
  };

  render() {
    return (
      <View>
        <RNCamera
          barCodeTypes={[RNCamera.Constants.BarCodeType.qr]}
          onBarCodeRead={() => console.log("barcode detected")}
          ref={cam => {
            this.camera = cam;
          }}
          style={{ height: 100, width: 100 }}
        >
          <View>
            <TouchableOpacity onPress={this.takePicture}>
              <Text>Take Photo</Text>
            </TouchableOpacity>
          </View>
        </RNCamera>

        <View />
      </View>
    );
  }
}

export default TakePicture;

```

### Expected behaviour
display console message after qrcode detection
qrcode state should be set with information

### Actual behaviour
1) on clicking take a picture, following displayed in the console
Path to image: file:///data/data/com.rnapp/cache/Camera/e4542c72-a629-4986-85e8-93aba7ca2d16.jpg
but no image is displayed in a mobile gallery.

2) also unable to detect qrcode

### Environment
- **Node.js - 9.5.0**:
- **React Native ^0.47.1**:
- **React Native platform - android 3.0.1**

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-03-16:
can u provide more information and stacktrace?

Marian0, 2018-03-16:
Same issue here. There is not stacktrace to paste since the camera simply does not scan any QR code.

No exceptions or errors throws.

I've read other issue about this and someone fixed it by changing the aspect ratio of the camera but it didn't work here.

```
Tested on:

Motorola G5S plus
Android 7.1.1

```

Thanks

Marian0, 2018-03-16:
#1258

jgfidelis, 2018-03-16:
```
on clicking take a picture, following displayed in the console
Path to image: file:///data/data/com.rnapp/cache/Camera/e4542c72-a629-4986-85e8-93aba7ca2d16.jpg
but no image is displayed in a mobile gallery.
```

Read the docs, it is the expected behavior.

The qr code part is a known bug.

ketankulkarni, 2018-03-17:
when this code is used for new RNCamera its not working

Marian0, 2018-03-18:
I've fixed it by changing the aspect ratio to another option.

DogsForArms, 2018-03-20:
RNCamera never fires (for android) seem like it will fire a barcode scan however Camera succeeds when i have aspect={Camera.constants.Aspect.fit}

sibelius, 2018-03-20:
use the right aspect ratio

closed 2018-03-20 by sibelius

K-Leon, 2018-03-20:
@sibelius  like i wrote - this won't work on every device. I tried it with every available aspect ratio. Even though old version works perfectly fine.

SimonErm, 2018-04-04:
#1422 Try master for qr-scan

izaa, 2018-04-07:
No help even using master branch. I need to restart the node server and then the first scan is ok, second one never comes.

SimonErm, 2018-04-09:
@izaa can you open a new issue with description, platform, used versions...? I think this is a new problem.

ketankulkarni, 2018-04-16:
QR code Scanning with RNCamera

```
import React, { Component } from "react";
import {
  StyleSheet,
  Dimensions,
  Text,
  View,
  Image,
  TouchableOpacity
} from "react-native";
import { RNCamera } from "react-native-camera";

class Scanner extends Component {
  constructor(props) {
    super(props);
    this.state = {
      qrcode: ""
    };
  }

  onBarCodeRead = e => {
    this.setState({ qrcode: e.data });
  };

  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          barCodeTypes={[RNCamera.Constants.BarCodeType.qr]}
          flashMode={RNCamera.Constants.FlashMode.on}
          style={styles.preview}
          onBarCodeRead={this.onBarCodeRead}
          ref={cam => (this.camera = cam)}
        >
          <Text
            style={{
              backgroundColor: "white"
            }}
          >
            {this.state.qrcode}
          </Text>
        </RNCamera>
      </View>
    );
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: "row"
  },
  preview: {
    flex: 1,
    justifyContent: "flex-end",
    alignItems: "center"
  }
});

export default Scanner;

```

avshenuk, 2018-04-16:
onBarCodeRead={this.onBarCodeRead}

I suppose here you need this.onBarCodeRead.bind(this)

ketankulkarni, 2018-04-16:
@avshenuk 
actually this is working code..
and no need to bind method as i have used arrow function

avshenuk, 2018-04-16:
Oh, arrow, right. Sorry :)

ducpt2, 2018-05-03:
@sibelius use the right aspect ratio => what do you mean? 
I try to use aspect={RNCamera.Constants.Aspect.fit} but it always has error.

IgorDePaula, 2019-02-06:
I copied the example and not working =/

xfyre, 2019-08-24:
Specifying `aspect: 1` actually solved this for me on RN 0.60.5 and RNCamera 3.3.0. But the fact that it's not even mentioned in the documentation is very confusing.

gabrieljoelc, 2020-05-14:
> Specifying `aspect: 1`

@xfyre are you referring to setting the `aspect` `prop` on the `RNCamera` component? something like:
```javascript
<RNCamera aspect={1} />
```

xfyre, 2020-05-14:
@gabrieljoelc Yes, that's what fixed the issue for me

andrefangeloni, 2020-05-26:
@xfyre On Android is working great, but on iOS is only working on iPhone 7 and iPhone 11. iPhone 6S and iPhone8/Plus aren't. Do you know what could be?