react-native-camera/react-native-camera #1360: Build Dependencies for android about face detection

hwangjr, 2018-03-17:
I am using the latest version 1.2.0, and I don't need to use face detection, I found that `android/app/build.gradle` has changed:
```
    compile (project(':react-native-camera')) {
    	exclude group: "com.google.android.gms"
	compile 'com.android.support:exifinterface:27.+'
	}
    compile ("com.google.android.gms:play-services-vision:10.2.0") {
    	force = true;
	}
```

I wonder if I just use the old dependencies (In case i don't need face detection), is it ok?
```
    compile project(':react-native-camera')
```

hwangjr, 2018-03-17:
maybe also can make android face detection optional.

jgfidelis, 2018-04-15:
I think on master branch it is now optional.  Try with:
```
compile (project(':react-native-camera')) {
    	exclude group: "com.google.android.gms"
	compile 'com.android.support:exifinterface:27.+'
	}
```

n1ru4l, 2018-09-25:
This seems to be solved. I am going to close this. If the the documentation is not clear enough, please send a pull request!

closed 2018-09-25 by n1ru4l