react-native-camera/react-native-camera #1361: RNCamera not saving photo to even Cache

Narutuffy, 2018-03-17:
### Which implementation are you using

*RNCamera* 

### Steps to reproduce
1. Use RNCamera to take photo.

This is the code for takeasyncphoto I am using:
```
function() {
        if (this.camera) {
          console.log("Hello");
          const options = { quality: 0.5, base64: true };
          this.camera.takePictureAsync(options).then(data => {
            ToastAndroid.show(data.uri, ToastAndroid.LONG);
            CameraRoll.saveToCameraRoll(data.uri);
            console.log(data.uri);
          });
        }
      }
```

### Expected behaviour
The cached image should be accessible

### Actual behaviour
It says the image is saved in ```file:///data/user/0/com.prank_selfie/cache/Camera/8f6a8888-db1a-4bfa-a047-d91586032d26.jpg``` but I cant see it anywhere. Also tried using (React Native) CameraRoll to save it to the camera roll, it still is not able to save it

### Using
"react-native": "0.54.1",
"react-native-camera": "^1.0.2",

jgfidelis, 2018-03-19:
Can you debug on native code why it is not saving on cache? Its weird, maybe a permission issue? Do you have all permissions set on your android manifest?

atifaziz1, 2018-03-23:
@jgfidelis same issue I am facing. All permissions are added in manifest.

jgfidelis, 2018-03-30:
You must debug in native code and see if it is really saving, etc

huyvu96, 2018-04-03:
same issue, but i did anything i don't still find a picture saving? Please help me

Narutuffy, 2018-04-03:
Could not find the exact location when it is saved to cache but when you save it to your gallery, the image is visible. To save it to your gallery use `CameraRoll.saveToCameraRoll(data.uri);`

closed 2018-04-03 by sibelius

atrauzzi, 2018-12-17:
Encountering this issue as well @jgfidelis -- what are the permissions I should have?

I've currently got:

```
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

Moseqi, 2019-10-18:
=> Open Device File Explorer in Android Studio
=> Open folder /data/data/yourpackagename/cache/Camera