react-native-camera/react-native-camera #1362: Toggling Camera on and off sometimes shows a black preview screen on iOS

kraffslol, 2018-03-18:
### Which implementation are you using
RNCamera

### Steps to reproduce
1. Render RNCamera based on a boolean state property.
2. Turn it on/off using a button.

### Expected behaviour
The preview should show as usual when the camera is on.

### Actual behaviour
The preview sometimes shows up as just a black screen, forcing you to turn it on and off until the preview shows up correctly again. The barcode scanner still works, but the preview doesnt.

### Environment
- **Node.js version**: 9.8.0
- **React Native version**: 0.53.3
- **React Native platform + platform version**: iPhone 7, iOS 11.2.6

### react-native-camera
**Version**: Master Commit 1778fbd

jgfidelis, 2018-03-18:
I think a solution could be to implement startPreview and stopPreview at the native code and call them when appropriate

kraffslol, 2018-03-19:
I made a quick implementation of startPreview and stopPreview to see if it would help, but it doesn't seem to be of much help in this specific case. There are still times where the camera will fail to render the preview at the "initialization" of the component. While calling startPreview afterwards did fix it, there still isn't a way to tell if the preview failed to render in the first place.

I suppose you could do something like setting up a setTimeout and call startPreview every X seconds incase the preview fails. But I think a better solution would be identifying why the preview fails to render in the first place. 😛

jgfidelis, 2018-03-19:
Agree with everything that you said. We are caught up in a big project and cannot help you with this bug right now, but this PR would be of great value

closed 2019-06-08 by stale[bot]