react-native-camera/react-native-camera #1363: android二维码扫描，扫描框中没有图像，是不是没有调用摄像头。权限已给。

wabgy, 2018-03-19:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RCTCamera

### Steps to reproduce
1.
2.
3.

### Expected behaviour
RN版本0.48.2 使用RCTCamera二维码扫描，无法获取画面。手电也不能打开。

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:0.48.2
- **React Native platform + platform version**:Android 6.0

### react-native-camera
**Version**: npm version

[image]

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-03-19:
抱歉

closed 2018-03-19 by jgfidelis

jgfidelis, 2018-03-19:
English please!