react-native-camera/react-native-camera #1364: Error. No camera found.

dann1609, 2018-03-19:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1. Use RCTCamera Or RNCamera.
2. Use the next code:
```
this.camera.capture()
                .then((data) => {
                  console.log('image captured: '+JSON.stringify(data));
                })
                .catch(err =>console.error(err));
```
3. Try on android 5.0.0
4. See console error.

### Expected behaviour
Take a photo

### Actual behaviour
No photo is taken

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: 0.12 and 1.0.2

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-03-19:
simulator or real device?

zachrnolan, 2018-07-27:
Seeing this for a few users on real devices.

react: 16.3.0-alpha.1
react-native: 0.53.3
react-native-camera: 0.13.0

jgcmarins, 2018-07-27:
@zachrnolan try version 1.1.5

n1ru4l, 2018-09-24:
Closed due to missing feedback. If you still experience this issue with the newest version open a new issue

closed 2018-09-24 by n1ru4l