react-native-camera/react-native-camera #1365: What is the correct syntax for using the quality property? (video)

arapocket, 2018-03-19:
```
                <RNCamera
                    ref={(cam) => {
                        this.camera = cam;
                    }}
                    style={styles.preview}
                    mirrorImage={false}
                    quality = {RNCamera.Constants.VideoQuality.480p}
                >
```

Using `quality = {RNCamera.Constants.VideoQuality.480p}` results in a syntax error at `.480p`

jgfidelis, 2018-03-23:
iOS code is:
```
@"VideoQuality": @{
                     @"2160p": @(RNCameraVideo2160p),
                     @"1080p": @(RNCameraVideo1080p),
                     @"720p": @(RNCameraVideo720p),
                     @"480p": @(RNCameraVideo4x3),
                     @"4:3": @(RNCameraVideo4x3),
                     },
```

So `RNCamera.Constants.VideoQuality.480p` should work.

Try to `console.log(RNCamera.Constants);` and see the object.

Your code looks correct.

closed 2018-03-23 by sibelius

astutetechnp, 2018-04-24:
Try this for Android
`const options = {maxDuration:5, quality:RNCamera.Constants.VideoQuality['480p']}`