react-native-camera/react-native-camera #1366: Incompatible block pointer types

chixpc123, 2018-03-20:
hello @sibelius please checkout my issue. 

### Expected behaviour
it should work fine in real device

### Actual behaviour
I can't run on device, it showing error "Incompatible block pointer types...." as you can show in attached snapshot

### Environment
- **Node.js version**: v8.7.0
- **React Native version**: 0.54.2
- **React Native platform + platform version**: iOS 9.2

### react-native-camera
**Version**: git+https://git@github.com/react-native-community/react-native-camera

[image]
[image]
[image]

Thanks.

jgfidelis, 2018-03-20:
Hmmm, maybe a problem with RN 0.54.2... they may have changed their method's signatures in the bridge library.

jgfidelis, 2018-03-20:
Looks like they changed view registry from a NSDictionary to a RCTSparseArray type from your print screens...

JeffreyHyer, 2018-07-11:
Any update on this? Running into the same issue on `react-native@0.54.4`

closed 2019-06-08 by stale[bot]