react-native-camera/react-native-camera #1367: RNCamera scan QR is fast, but the scan bar code is slow

hrh94, 2018-03-20:
The following is a part of the code：

```
import { RNCamera } from 'react-native-camera';
import { debounce } from 'throttle-debounce'; 

 state = {
    lightTurn: RNCamera.Constants.FlashMode.off,
  };

<RNCamera
          ref={ref => (this.camera = ref)}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={lightTurn}
          onBarCodeRead={debounce(2000, true, this.read)}
          permissionDialogTitle={'相机权限'}
          permissionDialogMessage={'我们需要使用您手机的相机权限'}
/>
```

hrh94, 2018-03-21:
IOS scan barcode is a bit slow, and Android scan can hardly sweep the barcode

hrh94, 2018-03-21:
I've solved this problem by Camera

```
import Camera from 'react-native-camera';
import { debounce } from 'throttle-debounce';

  state = {
    lightTurn: Camera.constants.TorchMode.off,
  };

<Camera
          ref={ref => (this.camera = ref)}
          style={styles.preview}
          torchMode={lightTurn}
          onBarCodeRead={debounce(2000, true, this.read)}
          barCodeTypes={
          [
            Camera.constants.BarCodeType.qr,
            Camera.constants.BarCodeType.ean13,
            Camera.constants.BarCodeType.upce,
          ]}
          permissionDialogTitle={'相机权限'}
          permissionDialogMessage={'我们需要使用您手机的相机权限'}
        />
```

jarnail1004, 2018-03-21:
@hrh94  
can you please add your code here. I am able to take pic but not able to scan barcode. Your code will help me to understand what i am missing.

closed 2018-03-27 by Fconstant

novpiseth, 2018-06-03:
@hrh94  can you share your full code of scanbarcode here? thanks you before hand. bacuse I'm kind of dummy coder. thanks for share.