react-native-camera/react-native-camera #1368: How to avoid multiple scans in one second

hrh94, 2018-03-20:
When I scan the bar code, I want it to take only one scan in two seconds, in fact it's going to keep scanning.

chen2584, 2018-03-23:
My current solution is
```javascript
onBarCodeRead = (scanResult) => {
    if(!this.barcodeList.includes(scanResult.data)) {
        this.barcodeList.push(scanResult.data);
    }
}
```

krystianpach, 2018-03-26:
You can use component state:

`
export class BarcodeScanner extends React.Component {
    state = {
        isBarcodeScannerEnabled: true
    }
    
    constructor(props) {
        super(props)
        
        this.onBarCodeRead = this.onBarCodeRead.bind(this)
    }
    
    onBarCodeRead(event) {
        if (this.state.isBarcodeScannerEnabled) {
            // code //
            alert(event.data)
            this.setState({
                isBarcodeScannerEnabled: false
            })
        }
    }
    
    render() {
        return (
            <RNCamera
                style={styles.preview}
                onBarcodeRead={this.onBarCodeRead}
            />
        )
    }
}
`

closed 2018-03-26 by sibelius

Bitenca, 2018-08-07:
Thanks @krystianpach! Like a charm!

prateekgoel, 2018-11-27:
Using component state doesn't help. As setState is asynchronous, there are many times when multiple scans are triggered before the state changes.
@sibelius Please reopen this issue. I think react-native-camera should provide a better solution.

KaoPengHsiang, 2019-02-26:
I have same problem when after setting isBarcodeScannerEnabled state, it take lots of scans in very short times that state didn't change in time. Is there any other solution can help?   > < "

chen2584, 2019-02-26:
@KaoPengHsiang This was my ugly solution. I never test it without `stopPreview` variable. If you do, please let me know the result.

```javascript
constructor() {
    this.stopPreview = false; // I don't save this in state, because setState() is asynchronous
}

onBarCodeRead = async(scanResult) => {
    if(!this.stopPreview) {
        this.stopPreview = true;
        this.refs.camera.stopPreview(); // force camera to stop preview (also stop detect barcode)
        
        // your logic goes here

        this.stopPreview = false;
        this.refs.camera.stopPreview(); // force camera to start preview again
    }
}

render() {
    return (
        <Camera ref="camera" onBarCodeRead={this.onBarCodeRead} />
    );
}
```

Jonnakutivijaykumar, 2019-07-03:
> I have same problem when after setting isBarcodeScannerEnabled state, it take lots of scans in very short times that state didn't change in time. Is there any other solution can help? > < "

Hello did u find any solution for this.

KaoPengHsiang, 2019-07-03:
@Jonnakutivijaykumar I finally use a ugly solution, too. You can try it, it works for me.
```js
i = 0;
onBarCodeRead = result => {
    this.i++;
    if (this.i++ == 1) {
//your logic here
    if (
        this.state.isBarcodeScannerEnabled &&
        Actions.currentScene !== "SecondPage"
      ) {        
        // code //
        this.setState({
          isBarcodeScannerEnabled: false
        });
     Actions.currentScene !== "SecondPage"   ? Actions.push("SecondPage", { url: data }) : {} ;
}
````

chengsam, 2020-10-07:
> Using component state doesn't help. As setState is asynchronous, there are many times when multiple scans are triggered before the state changes.
> @sibelius Please reopen this issue. I think react-native-camera should provide a better solution.

Just use a variable instead of state