react-native-camera/react-native-camera #1369: RNCamera crashes on ratio update (Android)

jpudysz, 2018-03-20:
### Which implementation are you using

*RNCamera* 

### Steps to reproduce
1. Use physical device eg. Galaxy Grand Prime
2. On `onCameraReady` get `getSupportedRatiosAsync` and use it to update state with ratio property (initially it's undefined)
3. You should get exception

### Expected behaviour
It should accept supported ratio and render camera

### Actual behaviour

[image]

### Environment
- **7.6.0**:
- **0.53.3**:
- **Android 5.1.1**

### react-native-camera
**Version**: 1.0.1

ShahoG, 2018-03-21:
Also on a samsung galaxy s7 edge: 
- can't scan barcode (EAN39/EAN7) at all (same code works in iOS)
- can't return anything from getSupportedRatiosAsync, getting 'undefined'

Fconstant, 2018-03-23:
Can you provide your piece of code? Are you installing from the `master` branch?

jpudysz, 2018-03-23:
I have fixed version in `package.json` to `1.0.1`
Code is pretty straightforward.

```javascript
constructor(props) {
     super(props)

     this.state = {
          // ...
          ratio: undefined
     }
}

async getCameraRatio() {
   const ratios = await this.cameraRef.getSupportedRatiosAsync()

   this.setState({
       ratio: ratios[ratios.length - 1]
   })
}

render() {
    return (
        <RNCamera
             ref={cam => this.cameraRef = cam}
             type={RNCamera.Constants.Type.back}
             style={styles.preview}
             flashMode={this.state.cameraFlashMode}
             ratio={this.state.ratio}
             onCameraReady={this.getCameraRatio}
         >
           {this.renderTopOverlay()}
           {this.renderBottomOverlay()}
        </RNCamera>
    )
}
```

Fconstant, 2018-03-23:
Oh, I've seen this too. I made a workaround, try this:
```jsx
constructor(props) {
    super(props);

    this.state = {
        // ...
        ratio: undefined,
        remountCamera: false,
    };
}

componentDidUpdate() {
    if (this.state.remountCamera) {
        setTimeout(() => {
            this.setState({ remountCamera: false })
        }, 500);
    }
}

async getCameraRatio() {
    if (!this.state.ratio && this.cameraRef) {
        const ratios = await this.cameraRef.getSupportedRatiosAsync();
        this.setState({
            ratio: ratios[ratios.length - 1],
            remountCamera: true,
        });
    }
}

render() {
    return (
        !this.state.remountCamera && (
            <RNCamera
                ref={cam => (this.cameraRef = cam)}
                type={RNCamera.Constants.Type.back}
                style={styles.preview}
                flashMode={this.state.cameraFlashMode}
                ratio={this.state.ratio}
                onCameraReady={this.getCameraRatio}
            >
                {this.renderTopOverlay()}
                {this.renderBottomOverlay()}
            </RNCamera>
        )
    );
}
```
What I did is remounting the Camera component if requested, and then after 500ms I render it again.
Don't forget to add `if (!this.state.ratio)` to `getCameraRatio`... Otherwise it will cause an infinite loop.

If this works, don't close this issue, because this is still a bug, and should be fixed.

ShahoG, 2018-03-26:
I've tried this exact code but I'm now getting:

> TypeError: Cannot read property 'ratio' of undefined
>     at Object.getCameraRatio$

Fconstant, 2018-03-26:
Oh, try this instead:
```jsx
getCameraRatio = async () => {
    (...)
}
```

jpudysz, 2018-04-25:
It's working fine with version `1.1.1`

closed 2018-04-25 by jpudysz

student9128, 2019-08-23:
just getRadio like this. it works fine with me.
```
   async _getRatio(){
        const ratios = await this.camera.getSupportedRatiosAsync()
        console.log('xxx', ratios)
        if(ratios){
            this.setState({
                camRatio:ratios[ratios.length-1]
            })
        }
    }

<RNCamera
...
  onCameraReady={()=>{
                        this._getRatio()
                    }}
/>
```