react-native-camera/react-native-camera #1370: Android build failed, failed for task ':react-native-camera:processReleaseResources'.

JakeRawr, 2018-03-21:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1. ./gradlew clean && ./gradlew assembleRelease
2.
3.

### Expected behaviour
Tell us what should happen
Android would build an apk

### Actual behaviour
Tell us what happens instead
Android build failed

### Environment
- **Node.js version**: v8.1.0
- **React Native version**: 0.53
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc Android

### react-native-camera
**Version**: npm version or "master"
v1.0.1
> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

rickcha, 2018-03-21:
same error occurred to me today.

JakeRawr, 2018-03-21:
yea, it literally just started to happen an hour ago.

So apparently when building, it was downloading new play-services-gcm files
```
Download https://maven.google.com/com/google/android/gms/play-services-vision/12.0.0/play-services-vision-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-base/12.0.0/play-services-base-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-basement/12.0.0/play-services-basement-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-vision-common/12.0.0/play-services-vision-common-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-vision-license/12.0.0/play-services-vision-license-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-tasks/12.0.0/play-services-tasks-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-base-license/12.0.0/play-services-base-license-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-basement-license/12.0.0/play-services-basement-license-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-vision-common-license/12.0.0/play-services-vision-common-license-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-tasks-license/12.0.0/play-services-tasks-license-12.0.0.pom
Download https://maven.google.com/com/android/support/support-v4/26.1.0/support-v4-26.1.0.pom
Download https://maven.google.com/com/android/support/support-compat/26.1.0/support-compat-26.1.0.pom
Download https://maven.google.com/com/android/support/support-media-compat/26.1.0/support-media-compat-26.1.0.pom
Download https://maven.google.com/com/android/support/support-core-utils/26.1.0/support-core-utils-26.1.0.pom
Download https://maven.google.com/com/android/support/support-core-ui/26.1.0/support-core-ui-26.1.0.pom
Download https://maven.google.com/com/android/support/support-fragment/26.1.0/support-fragment-26.1.0.pom
Download https://maven.google.com/android/arch/lifecycle/runtime/1.0.0/runtime-1.0.0.pom
Download https://maven.google.com/android/arch/lifecycle/common/1.0.0/common-1.0.0.pom
Download https://maven.google.com/android/arch/lifecycle/common/1.0.0/common-1.0.0.jar
Download https://maven.google.com/android/arch/lifecycle/runtime/1.0.0/runtime-1.0.0.aar
Download https://maven.google.com/com/android/support/support-compat/26.1.0/support-compat-26.1.0.aar
Download https://maven.google.com/com/android/support/support-media-compat/26.1.0/support-media-compat-26.1.0.aar
Download https://maven.google.com/com/android/support/support-core-utils/26.1.0/support-core-utils-26.1.0.aar
Download https://maven.google.com/com/android/support/support-core-ui/26.1.0/support-core-ui-26.1.0.aar
Download https://maven.google.com/com/android/support/support-fragment/26.1.0/support-fragment-26.1.0.aar
Download https://maven.google.com/com/android/support/support-v4/26.1.0/support-v4-26.1.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-basement-license/12.0.0/play-services-basement-license-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-basement/12.0.0/play-services-basement-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-tasks-license/12.0.0/play-services-tasks-license-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-tasks/12.0.0/play-services-tasks-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-base-license/12.0.0/play-services-base-license-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-base/12.0.0/play-services-base-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-vision-common-license/12.0.0/play-services-vision-common-license-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-vision-common/12.0.0/play-services-vision-common-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-vision-license/12.0.0/play-services-vision-license-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-vision/12.0.0/play-services-vision-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-gcm/12.0.0/play-services-gcm-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-iid/12.0.0/play-services-iid-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-gcm-license/12.0.0/play-services-gcm-license-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-iid-license/12.0.0/play-services-iid-license-12.0.0.pom
Download https://maven.google.com/com/google/android/gms/play-services-iid-license/12.0.0/play-services-iid-license-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-iid/12.0.0/play-services-iid-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-gcm-license/12.0.0/play-services-gcm-license-12.0.0.aar
Download https://maven.google.com/com/google/android/gms/play-services-gcm/12.0.0/play-services-gcm-12.0.0.aar
```

then the error occurred

manask88, 2018-03-21:
same thing happening to me. I was on 0.13.0 though, but it seems like this may be affecting all versions. I tried adding  exclude group: "com.google.android.gms" to all libraries that had a dependency on com.google.android.gms, but still didn't work

JakeRawr, 2018-03-21:
Well...

so I removed react-native-camera from my app, tried to build it. But still getting the error.
```
:app:processReleaseResources FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processReleaseResources'.
> Error: more than one library with package name 'com.google.android.gms.license'
```
This time, it's `:app`.

So sounds like we just have to wait for Google to update this then.

sibelius, 2018-03-21:
Fix ur build.gradle check docs

closed 2018-03-21 by sibelius

PrimeObjects, 2018-03-21:
This issue is not closed yet.

// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    repositories {
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.2.+'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven { url "https://jitpack.io" }
        maven { url 'https://maven.google.com' }
    }
}

ilsonlasmar, 2018-03-21:
I'm having the same problem. My application started with the same error. I have not made any changes.

internet5, 2018-03-21:
Same error, I have not changed anything

rickcha, 2018-03-21:
@sibelius I don't know how genius you are, but reopen this ticket if you got no clue how to fix it.

dolphinflow86, 2018-03-21:
Same error here. I didn't change any code.

AdamChrist, 2018-03-21:
Same error

internet5, 2018-03-21:
Anybody tell me how to solve this?

AdamChrist, 2018-03-21:
change `node_modules/react-native-camera/android/build.gradle` :
` compile 'com.google.android.gms:play-services-vision:11.8.0'`
it can temporary fix to the problem

quytangvn, 2018-03-21:
Encounter the same issue. Why did someone close it? I've just intended to open the same issue.

PrimeObjects, 2018-03-21:
Hi AdamChrist, 

I did the same but I still got the same error. Do you mind to share your build.grade.

Thanks lot

internet5, 2018-03-21:
@AdamChrist Thanks, I can run it now！

PrimeObjects, 2018-03-21:
Hi AdamChrist,

Good finding. Thank you very much!

For any app that may depend on different version, please use the command below to find the 
dependency. 

$ ./gradlew -q app:dependencies

Cheers
Gary

AdamChrist, 2018-03-21:
@PrimeObjects  `node_modules/react-native-camera/android/build.gradle`
```
buildscript {
  repositories {
    jcenter()
    maven {
      url 'https://maven.google.com'
    }
  }

  dependencies {
    classpath 'com.android.tools.build:gradle:3.0.0'
  }
}

apply plugin: 'com.android.library'

android {
  compileSdkVersion 26
    buildToolsVersion "26.0.2"

  defaultConfig {
    minSdkVersion 16
    targetSdkVersion 26
    versionCode 1
    versionName "1.0.0"
  }
  lintOptions {
    abortOnError false
    warning 'InvalidPackage'
  }
}

repositories {
  mavenCentral()
  maven {
   url 'https://maven.google.com'
  }
  maven { url "https://jitpack.io" }
}

dependencies {
  compile 'com.facebook.react:react-native:+'
  compile "com.google.zxing:core:3.2.1"
  compile "com.drewnoakes:metadata-extractor:2.9.1"
  compile 'com.google.android.gms:play-services-vision:11.8.0'
  compile "com.android.support:exifinterface:+"

  compile 'com.github.react-native-community:cameraview:cc47bb28ed2fc54a8c56a4ce9ce53edd1f0af3a5'
}
```

PrimeObjects, 2018-03-21:
@AdamChrist I just noticed you was talking about node_modules/react-native-camera/android/build.gradle instead of the file in my project. It did fix the issue. 

Thanks again!
Gary

ilsonlasmar, 2018-03-21:
Thanks @AdamChrist . It worked!

brunsy, 2018-03-21:
Another temporary fix is to override it from your build.gradle
```
subprojects { subproject ->
    afterEvaluate{
      if(subproject.name == "react-native-camera" ){
        dependencies {
          compile ("com.google.android.gms:play-services-vision:11.8.0") {
            force = true;
          }
        }
      }
    }
}
```

JakeRawr, 2018-03-21:
If the above solution doesn't work for you, 

check other dependencies. I am using "react-native-device-info" and "react-native-push-notification", they also need to be changed

david1820, 2018-03-21:
I have the same problem only works when i manually change the `node_modules/react-native-camera/android/build.gradle`

sibelius, 2018-03-21:
Force can fix this

quytangvn, 2018-03-21:
@sibelius try it yourself. It doesn't work at all!

Fox-3394, 2018-03-21:
@quytangvn The problem has been solved, add config in android/build.gradle.
repositories {
        configurations.all {
            resolutionStrategy {
               // Add force (10.0.0 is version you want to use)
                force 'com.google.firebase:firebase-messaging:10.0.1' // If you using fcm
                force 'com.google.firebase:firebase-core:10.0.1' // If you using fcm
                force 'com.google.android.gms:play-services-gcm:10.0.1' // If you using fcm
                force 'com.google.android.gms:play-services-vision:10.0.1'
            }
        }

krystiansliwa, 2018-03-21:
@Fox-3394 Can you post here whole content of android/build.gradle?

sibelius, 2018-03-21:
```
compile ("com.google.android.gms:play-services-vision:11.8.0") {
            force = true;
          }
```

works

GreenCame, 2018-03-21:
I tried all solutions here only @Fox-3394 works ! thanks

Traviskn, 2018-03-21:
I've also tried all of the solutions here, but I'm still running into the same error:
```
Execution failed for task ':react-native-camera:processReleaseResources'.
> Error: more than one library with package name 'com.google.android.gms.license'
```

This is the current state of my gradle files:

`android/build.gradle`
```gradle
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    repositories {
        jcenter()
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.2.3'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
        maven {
            url 'https://jitpack.io'
        }
        configurations.all {
            resolutionStrategy {
                force 'com.google.android.gms:play-services-gcm:11.8.0'
            }
        }
    }
}
```

"dependencies" section of `android/app/build.gradle`
```gradle
dependencies {
    compile fileTree(include: ['*.jar'], dir: 'libs')
    compile 'com.android.support:appcompat-v7:26.0.2'
    compile 'com.facebook.react:react-native:+'
    // From node_modules
    compile project(':react-native-keep-awake')
    compile project(':react-native-device-info')
    compile project(':react-native-wheel-picker-android')
    compile project(':react-native-fetch-blob')
    compile project(':react-native-vector-icons')
    compile project(':react-native-sensitive-info')
    compile project(':react-native-i18n')
    compile project(':react-native-threads')
    compile project(':react-native-sqlite-storage')
    compile (project(':react-native-camera')) {
      exclude group: 'com.google.android.gms'
      compile 'com.android.support:exifinterface:27.+'
    }
    compile ('com.google.android.gms:play-services-vision:11.8.0') {
        force = true;
    }
}
```

Am I missing something?  I'd appreciate any tips from others here who got their build working 🙇

hungdev, 2018-03-21:
i get same issue.

hungdev, 2018-03-21:
thanks @AdamChrist it worked.

johnbodenhamer, 2018-03-21:
None of the above options worked for me. In my apps build.gradle, I had specific versions listed for things like play-services vision so it read `compile ("com.google.android.gms:play-services-vision:10.2.0") {
        force = true;
    }`

I guess Google APIs updated yesterday so changing the 10.2.0 to 12.0.0 worked for me. To avoid this in the future where I'll likely forget, we changed the version number to a + sign. 

`compile ("com.google.android.gms:play-services-vision:+") {
        force = true;
    }`

That seemed to work

vosmario, 2018-03-21:
configurations.all {
            resolutionStrategy {
                force 'com.google.android.gms:play-services-gcm:11.8.0'
                force 'com.google.android.gms:play-services-vision:11.8.0'
            }
        }

vosmario, 2018-03-21:
This work for me. In Android/build.gradle 
`allprojects {
    repositories {
        ...
        configurations.all {
            resolutionStrategy {
                force 'com.google.android.gms:play-services-gcm:11.8.0'
                force 'com.google.android.gms:play-services-vision:11.8.0'
            }
        }
    }
}
`

cooclsee, 2018-03-22:
Thanks @vosmario it worked.

formated code in `android/build.gradle` 
```diff
allprojects {
    repositories {

        // ...

+        configurations.all {
+          resolutionStrategy {
+            force 'com.google.android.gms:play-services-vision:11.8.0'
+            force 'com.google.android.gms:play-services-gcm:11.8.0'
+          }
+        }
    }

}

```

Traviskn, 2018-03-22:
@vosmario @cooclsee That config works for me!  I had config for `play-services-gcm`, but not for `play-services-vision`.  I noticed that a different library I had installed, `react-native-device-info` seemed to be playing into the dependency issue.  So it seems the reason some of these solutions don't work for all projects is because you have to take all of your other dependencies into account

pacozaa, 2018-03-23:
@Traviskn Mine seems to be `react-native-camera` that cause problem

Can I see your `android/app/build.gradle` ?

Traviskn, 2018-03-23:
@pacozaa This is my `android/app/build.gradle` dependencies
```gradle
dependencies {
    compile fileTree(include: ['*.jar'], dir: 'libs')
    compile 'com.android.support:appcompat-v7:26.0.2'
    compile 'com.facebook.react:react-native:+'
    // From node_modules
    compile project(':react-native-keep-awake')
    compile project(':react-native-device-info')
    compile project(':react-native-wheel-picker-android')
    compile project(':react-native-fetch-blob')
    compile project(':react-native-vector-icons')
    compile project(':react-native-sensitive-info')
    compile project(':react-native-i18n')
    compile project(':react-native-threads')
    compile project(':react-native-sqlite-storage')
    compile (project(':react-native-camera')) {
      exclude group: 'com.google.android.gms'
    }
    compile ('com.google.android.gms:play-services-vision:11.8.0') {
      force = true
    }
}
```

and this is my `android/build.gradle`
```gradle
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    repositories {
        jcenter()
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.2.3'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
        maven {
            url 'https://jitpack.io'
        }
        configurations.all {
          resolutionStrategy {
            force 'com.google.android.gms:play-services-vision:11.8.0'
            force 'com.google.android.gms:play-services-gcm:11.8.0'
          }
        }
    }
}
```

pacozaa, 2018-03-24:
@Traviskn Thank you hope the issue will be solved soon.

pacozaa, 2018-03-24:
@Traviskn Now my `mauron85/react-native-background-geolocation` error instead.

```
Execution failed for task ':react-native-mauron85-background-geolocation:processReleaseResources'.
> com.android.ide.common.process.ProcessException: Failed to execute aapt
```

but if I delete, 

```
configurations.all {
 resolutionStrategy {
   force 'com.google.android.gms:play-services-vision:10.2.0'
   force 'com.google.android.gms:play-services-gcm:10.2.0'
 }
}
```

I got error on `react-native-camera` back

How can they work together!?

pacozaa, 2018-03-26:
Ok, I am using this to substitute `react-native-camera`
https://github.com/wix/react-native-camera-kit

marvisamps, 2018-06-07:
@brunsy solution works like a charm for me! Thanks, bro!

brunsy, 2018-06-07:
@marvisamps In most cases you will just want to set it through your project-wide gradle properties as these are respected by react-native-camera now...

Manmohan-R, 2019-12-23:
> Configure project :app
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-camera-kit
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-gesture-handler
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-maps
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-reanimated
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-vector-icons
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

WARNING: The specified Android SDK Build Tools version (26.0.2) is ignored, as it is below the minimum supported version (28.0.3) for Android Gradle Plugin 3.5.2.
Android SDK Build Tools 28.0.3 will be used.
To suppress this warning, remove "buildToolsVersion '26.0.2'" from your build.gradle file, as each version of the Android Gradle Plugin now has a default version of the build tools.

> Task :react-native-camera-kit:compileDebugJavaWithJavac FAILED
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/CameraViewManager.java:10: error: package android.support.annotation does not exist
import android.support.annotation.ColorInt;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/CameraViewManager.java:11: error: package android.support.annotation does not exist
import android.support.annotation.IntRange;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/CameraView.java:5: error: package android.support.annotation does not exist
import android.support.annotation.ColorInt;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/barcode/BarcodeScanner.java:8: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/barcode/BarcodeScanner.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/barcode/BarcodeFrame.java:9: error: package android.support.annotation does not exist
import android.support.annotation.ColorInt;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/permission/CameraPermission.java:5: error: cannot find symbol
import android.support.v4.app.ActivityCompat;
                             ^
  symbol:   class ActivityCompat
  location: package android.support.v4.app
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/permission/CameraPermission.java:6: error: package android.support.v4.content does not exist
import android.support.v4.content.PermissionChecker;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/gallery/GalleryView.java:5: error: package android.support.v7.widget does not exist
import android.support.v7.widget.GridLayoutManager;
                                ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/gallery/GalleryView.java:6: error: package android.support.v7.widget does not exist
import android.support.v7.widget.RecyclerView;
                                ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/gallery/GalleryView.java:10: error: cannot find symbol
public class GalleryView extends RecyclerView {
                                 ^
[... 366 lines cut ...]
                ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/gallery/GalleryViewManager.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
92 errors

FAILURE: Build failed with an exception.

[... cut ...]
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 0s
11 actionable tasks: 1 executed, 10 up-to-date
error Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/getting-started.html

Manmohan-R, 2019-12-23:
Please give any solution about this 

> Configure project :app
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-camera-kit
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-gesture-handler
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-maps
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-reanimated
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

> Configure project :react-native-vector-icons
WARNING: The following project options are deprecated and have been removed: 
android.databinding.enableV2
Databinding v1 is removed.

WARNING: The specified Android SDK Build Tools version (26.0.2) is ignored, as it is below the minimum supported version (28.0.3) for Android Gradle Plugin 3.5.2.
Android SDK Build Tools 28.0.3 will be used.
To suppress this warning, remove "buildToolsVersion '26.0.2'" from your build.gradle file, as each version of the Android Gradle Plugin now has a default version of the build tools.

> Task :react-native-camera-kit:compileDebugJavaWithJavac FAILED
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/CameraViewManager.java:10: error: package android.support.annotation does not exist
import android.support.annotation.ColorInt;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/CameraViewManager.java:11: error: package android.support.annotation does not exist
import android.support.annotation.IntRange;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/CameraView.java:5: error: package android.support.annotation does not exist
import android.support.annotation.ColorInt;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/barcode/BarcodeScanner.java:8: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/barcode/BarcodeScanner.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/barcode/BarcodeFrame.java:9: error: package android.support.annotation does not exist
import android.support.annotation.ColorInt;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/permission/CameraPermission.java:5: error: cannot find symbol
import android.support.v4.app.ActivityCompat;
                             ^
  symbol:   class ActivityCompat
  location: package android.support.v4.app
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/camera/permission/CameraPermission.java:6: error: package android.support.v4.content does not exist
import android.support.v4.content.PermissionChecker;
                                 ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/gallery/GalleryView.java:5: error: package android.support.v7.widget does not exist
import android.support.v7.widget.GridLayoutManager;
                                ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/gallery/GalleryView.java:6: error: package android.support.v7.widget does not exist
import android.support.v7.widget.RecyclerView;
                                ^
/Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/gallery/GalleryView.java:10: error: cannot find symbol
[... 368 lines cut ...]
                ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/manmohan.r/Desktop/MainProject1/ABFRL-VH/vanHeusen/node_modules/react-native-camera-kit/android/src/main/java/com/wix/RNCameraKit/gallery/GalleryViewManager.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
92 errors

FAILURE: Build failed with an exception.

* What went wrong:
[... cut ...]
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 0s
11 actionable tasks: 1 executed, 10 up-to-date
error Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/getting-started.html

wayferer, 2020-02-03:
Adding this to my `android/build.gradle` helped

```
subprojects {
    afterEvaluate {subproject ->
        if((subproject.plugins.hasPlugin('android') || subproject.plugins.hasPlugin('android-library'))) {
            android {
                compileSdkVersion rootProject.ext.compileSdkVersion
                buildToolsVersion rootProject.ext.buildToolsVersion
                lintOptions {
                    abortOnError false
                }
            }
        }
    }
}
```

yogendrakumar96, 2021-03-04:
none of the solution solved my issue as it is saying same error.
> Task :react-native-camera:compileGeneralDebugJavaWithJavac FAILED
can anyone help me out