react-native-camera/react-native-camera #1371: file not found

pohsiu, 2018-03-21:
I want to use faceDetection function,
But got this error
[image]

so I pod install to link & add frameworks myself

> Podfile

`target 'camera' do`
`pod 'react-native-camera', subspecs: ['RCT', 'RN', 'FaceDetector'], path: '../node_modules/react-native-camera'`
`end`

> Add to framework(by 2 way, both failed and get same error) 

1.
[image]
2. (this way will add Link Binary With Libraries auto)
[image]

pod install warning (no idea how to fix this, either)
[image]

### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**: v8.9.1
- **React Native version**: "0.53.3"
- **React Native platform + platform version**: iOS 11.2

### react-native-camera
**Version**: npm version or "master" latest version(2018/03/19 installed) from git+https://git@github.com/react-native-community/react-native-camera

zhaiyjgithub, 2018-03-21:
same issue.

jgfidelis, 2018-03-22:
can you try master branch?

jgfidelis, 2018-04-07:
Is the bug still happening?

n1ru4l, 2018-09-24:
Closed due missing feedback. Please feel free to reopen the issue if the issue does still occur.

closed 2018-09-24 by n1ru4l