react-native-camera/react-native-camera #1373: Promise does not return

chenxiong666, 2018-03-21:
[image]

[image]
I've tried these two ways no return value

jgfidelis, 2018-03-21:
is `this.camera` different than null or undefined, is takePictureAsync being called?

chenxiong666, 2018-03-22:
[image]
but now,I have this problem

chenxiong666, 2018-03-22:
[image]
I`m directly executing your demo.

chenxiong666, 2018-03-22:
I have found that sometimes I can get the path but sometimes it doesn't

jgfidelis, 2018-03-22:
Android? iOS?

chenxiong666, 2018-03-23:
android,I found it normal to pack APK

kasterlod, 2018-03-26:
The same for me. Seems like I can take picture again after clear app data. Is there any way to clear library cache or smth?

@jgfidelis @chenxiong666

kasterlod, 2018-03-26:
there's should be possibility to clear cache

tonilxm, 2018-03-27:
I just encounter this problem as well. happen on android device (not simulator).

tonilxm, 2018-03-27:
@jgfidelis : takePictureAsync is being called. promise never return anything.

nick-creston, 2018-03-27:
I encountered a similar problem
Function takePictureAsync did not return anything
After adding
`<uses-permission android: name = "android.permission.CAMERA" />`
in app/src/main/AndroidManifest.xml earned

Sangrene, 2018-03-27:
@kasterlod if you need to clear the cache you can use this library : https://github.com/midas-gufei/react-native-clear-app-cache .

jgfidelis, 2018-03-27:
@nick-creston 

After adding the permission, it worked?

Do you guys with the issues have all the necessary permissions specified in the docs?

nick-creston, 2018-03-28:
@jgfidelis 

In my case, it worked. 
I use device for development builds: Samsung E5 (Android 5.1.1)

yz-xlame, 2018-04-09:
takePictureAsync and recordAsync promise never return anything.

ketankulkarni, 2018-04-14:
takePicture() {
    if (this.camera) {
      this.camera
        .takePictureAsync()
        .then(data => {
          CameraRoll.saveToCameraRoll(data.uri, "photo");
          console.log(data);
        })
        .catch(err => {
          alert("err");
          console.log(err);
        });
    }
  }

ketankulkarni, 2018-04-14:
@yz-xlame takepictureAsync return uri,height,width
but after takepictureAsync it take time upto 4-5 sec

stueynet, 2018-05-10:
Same issue here. But seems pretty intermittent. Sometimes it works and other times the promise never returns. We already have the permissions in there. Anyone else have an idea of how to get this working?

pstanton, 2018-05-31:
clean gradle, and re-run. seemed to work for me:
```
cd android
gradlew clean
cd ..
react-native run-android
```

felansu, 2018-06-10:
Thanks @pstanton for temporal solution 👍

CptFabulouso, 2018-06-19:
In my case it happens because [doInBackground()](https://github.com/react-native-community/react-native-camera/blob/8e1a71dc286cb8bcaa6e3a41e0cefd7c59d492cc/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java#L48) is not called on android, which should be called with 
[new ResolveTakenPictureAsyncTask(data, promise, options, cacheDirectory).execute()](https://github.com/react-native-community/react-native-camera/blob/8e1a71dc286cb8bcaa6e3a41e0cefd7c59d492cc/android/src/main/java/org/reactnative/camera/RNCameraView.java#L93). 
Could replacing the `.execute()` with `.executeOnExecutor(AsyncTask.THREAD_POOL_EXECUTOR)` help? [See this issue](https://stackoverflow.com/a/26890971) 

Right now it seems to fix the issue for me, but even before this issue happened randomly, so can't say for sure yet.

CptFabulouso, 2018-07-11:
It did, patch submitted

harrigee, 2018-09-28:
For me it was an issue because of a conditional rendering of the camera component. When I was trying to create the image and then showed the ActivityIndicator, instead of the Camera component, the Promise never resolved. On iOS this was not an issue (probably a race condition).

harrigee, 2018-10-09:
@slorber Can you play around with the rendering order? I would try to place the Camera Component more upwards in the dom. As you can see, it is the first element inside the container in my case. This worked.

`<View style={styles.container}>
                <RNCamera
                    ref={(ref: any) => {
                        this.camera = ref;
                    }}
                    style={{ flex: 1 }}
                    type={RNCamera.Constants.Type.back}
                    flashMode={RNCamera.Constants.FlashMode.off}
                    permissionDialogTitle={'Permission to use camera'}
                    permissionDialogMessage={'We need your permission to use your camera phone'}
                    onBarCodeRead={(barcode) => {
                        console.log(barcode)
                    }}
                />
                {this.state.creatingPhoto ?
...
`

ksairi, 2018-11-08:
any solid solutions about this?

OyeOvi, 2018-11-21:
Anything on this yet?
takePictureAsync never returns!

ksairi, 2018-11-21:
@OyeOvi  make sure you don't any spinning loader or related. I removed mine and takePictureAsync started to return!

OyeOvi, 2018-11-21:
@ksairi there is no loader or anything like that
```
snap = async () => {
    if (this.camera) {
      const { uri } = await this.camera.takePictureAsync();
      this.setState({ image: uri });
      const { faces } = await this.detectFaces(uri);
      this.setState({ faces });
    }
  };
```

ksairi, 2018-11-21:
@OyeOvi  try to no to set the state, neither before nor after ``takePictureAsync``. console.log the uri and see if it returns.

OyeOvi, 2018-11-21:
Tried doing so, same thing, I'm running it in Android using Expo

OyeOvi, 2018-11-21:
Figured it out!
There were some caches issue I guess - because same code was working fine in [Expo Smack](https://snack.expo.io/)
I cleared the data of Expo App and it stated to work fine!

Yajan, 2019-01-09:
I am not getting any response from `this.camera.capture`  

  
    takePicture() {
        console.log("In Take picture");
        const options = {}
        this.camera.capture({metadata: options})
            .then(data => {
                console.log("captured Image data",data);
                CameraRoll.saveToCameraRoll(data.uri, "photo");
            })
            .catch(err => {
                console.error('capture picture error', err);
            });
    }

    render() {
        return (
            <View >
                    <Camera
                        captureTarget={Camera.constants.CaptureTarget.disk}
                        ref={cam => {
                            this.camera = cam;
                        }}
                        style={styles.view}
                        aspect={Camera.constants.Aspect.fill}
                        captureTarget={Camera.constants.CaptureTarget.disk}
                    >
                        <Text
                            style={styles.capture}
                            onPress={this.takePicture.bind(this)}
                        >
                            CAPTURE
                        </Text>
                    </Camera>
            </View> 
        );
    }

In androidManifest.xml I have 

    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

n1ru4l, 2019-01-09:
@Yajan RCTCamera is deprecated. Please use RNCamera instead.