react-native-camera/react-native-camera #1375: Execution failed for task ':react-native-camera:processReleaseResources'. more than one library with package name 'com.google.android.gms.license'

jinxin479, 2018-03-21:
### Warning
Execution failed for task ':react-native-camera:processReleaseResources'. 
more than one library with package name 'com.google.android.gms.license'

### 

react-native-camera  master branch 

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
           url 'https://maven.google.com'
       }
        maven {
            url 'https://jitpack.io'
        }
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
    }
}

buildToolsVersion "26.0.1"

if I remove  
maven {
           url 'https://maven.google.com'
       }
error 👎 
could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:appcompat-v7:26.0.1.

anandwahed, 2018-03-21:
@jinxin479 Seems to be a duplicate of #1370 please check this [comment](https://github.com/react-native-community/react-native-camera/issues/1370#issuecomment-374830357) for the fix

hippycore, 2018-03-21:
@jinxin479 this [comment](https://github.com/react-native-community/react-native-camera/issues/1370#issuecomment-374868773) helped me

sibelius, 2018-03-21:
avoid version 12 for now

closed 2018-03-21 by sibelius