react-native-camera/react-native-camera #1376: Error: more than one library with package name 'com.google.android.gms.license'

jarnail1004, 2018-03-21:
### Error
:react-native-camera:processReleaseResources FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:processReleaseResources'.
> Error: more than one library with package name 'com.google.android.gms.license'

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 1 mins 5.5 secs
Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/getting-started.html

### Steps to reproduce
1. Initialise app with **react-native init** bar in directory E:\scanner
   - it works well on physical device (Android phone) after starting android-studio and attaching phone with USB. Works well till here
2. installed camera package as in [link](   https://github.com/react-native-community/react-native-camera#mostly-automatic-install-with-react-native
):
npm install react-native-camera --save
react-native link react-native-camera
3.run finally **react-native run-android** and above error produced.

### Expected behaviour
It should work smoothly because i didnt call camera in code yet.

### Actual behaviour
Build failed and produce above error. 

### Environment
- **Node.js version**:8.10.0
- **React Native version**: 0.54.2
- **React Native version CLI**: 2.0.1
- **React Native platform + platform version**: Windows 10 pro, Android 7.0, etc

gorhgorh, 2018-03-21:
I'm having the same issue after cloning a working camera project somewhere else using "react-navigation": "1.2.0"

conilas, 2018-03-21:
go to node-modules/react-native-camera/android and edit build.gradle

change from:

dependencies{ .... compile 'com.google.android.gms:play-services-vision:+ }

to:

dependencies{ .... compile 'com.google.android.gms:play-services-vision:11.8.0 }

as a temporary workaround

btoueg, 2018-03-21:
Same issue seems to happen on every react native dependency:

 - https://github.com/evollu/react-native-fcm/issues/857
 - https://github.com/facebook/react-native/issues/18479
 - ...

sibelius, 2018-03-21:
`compile ('com.google.android.gms:play-services-vision:11.8.0) { force: true }`

works as well

closed 2018-03-21 by sibelius

adrianolsk, 2018-03-22:
@sibelius that did't work here, even forcing the version this way I got the same error.

The only thing that worked is setting this in `android/build.gradle`

```

allprojects {
    repositories {
       configurations.all {
            resolutionStrategy {
                force 'com.google.android.gms:play-services-vision:11.8.0'
            }
        }
}
```