react-native-camera/react-native-camera #1377: Front camera recorded video is upside down

cristiantone, 2018-03-21:
Hi,

I have an issue with video recording with the front camera. The video is recorded upside down although the preview is the right way. The recording for the back camera works perfect.
I've been trying to fix this for the last 3 days. Are there any solutions?

### Which implementation are you using

RNCamera

### Expected behaviour
Record a video with the front camera of the phone and then play it.

### Actual behaviour
The video is recorded but when I played it is upside down although the preview was the right way.

### Environment
- **Node.js version**: v4.2.6
- **React Native version**: 0.52.0
- **React Native platform + platform version**: Android 8.0, Android 5.0, etc

### react-native-camera
**Version**: npm version ^1.0.2

condorman, 2018-03-21:
Hi, i find the solution with EXPO developer, the problem is on cameraview repository, I pull this change: https://github.com/react-native-community/cameraview/pull/3

jgfidelis, 2018-03-22:
Great work!

closed 2018-03-22 by jgfidelis

condorman, 2018-03-22:
@jgfidelis, we need the update of android/build.gradle with:

compile 'com.github.react-native-community:cameraview:d529251d24c0a367a28eea28f0eac2269d12f772'

thx!!!

sibelius, 2018-03-22:
@condorman can u send a PR to upgrade this?