react-native-camera/react-native-camera #1378: If you want to make any visual or audio effect, there is no way to understand when an image is taken. The response that comes from promise is too late. For example try it with flash mode on and you will see it.  Are you planning to have more event triggers?

madebytilde, 2018-03-21:
If you want to make any visual or audio effect, there is no way to understand when an image is taken. The response that comes from promise is too late. For example try it with flash mode on and you will see it.  Are you planning to have more event triggers?

sibelius, 2018-03-21:
we could have more events?

which events do u wanna?

madebytilde, 2018-03-21:
when picture is taken and flash light is fired.

sibelius, 2018-03-21:
could u send a Pr so we can make sure this makes sense?

madebytilde, 2018-03-21:
Sorry ... we do not have Android or iOS developers in our team. I do not know if I'm clear ... but if you open the camera and click the button to take a picture, there is no indicator that shows that the job is done (no audio or visual indicator). How can I know when the image is taken with RNCamera , if no event has been fired… response from promise is too late.

madebytilde, 2018-03-21:
I think you need to fire many more events... for example onCameraReady should be toggle, when you click the button to switch the camera from front to back because during this time the camera is not ready… but it is completely frozen... More events are not a bad thing… More events will make this component much more usable.

jgfidelis, 2018-03-21:
`How can I know when the image is taken with RNCamera , if no event has been fired… response from promise is too late.`

This does not make sense, in my opinion. The promise is resolved when the picture is taken and ready. Before that, there is no picture. Response from promise is not too late, it is the earliest possible.

madebytilde, 2018-03-21:
No, because it takes a few ms to save the picture on cache directory… while flash light is fired. So we do not have the ability to make any animated effect when the picture is taken… because the visual effect should occur when flash light is fired not after the picture is saved on cache directory and we have uri.

closed 2019-06-08 by stale[bot]