react-native-camera/react-native-camera #1379: Unable to take a photo on Samsung Galaxy Grand Prime

jpudysz, 2018-03-21:
### Which implementation are you using
*RCTCamera*, cannot take a photo on *RNCamera* because of #1369 

### Steps to reproduce
1. Use older Android like Samsung Galaxy Grand Prime
2. Use RCTCamera
3. Take a photo

### Expected behaviour
Should return a photo

### Actual behaviour
Crash in native code occured

### Environment
- **Node.js version**: 7.6.0
- **React Native version**: 0.53.3
- **Android**: 5.1.1

### react-native-camera
**Version**: 1.0.1

### Stack trace
[image]

### Additional info
I'm mounting/unmounting camera multiple times because of specific use case (full screen photo preview). I was trying to preserve camera ref and use `stopPreview` with `startPreview` but with no luck (I don't know if it's even related).

deehuey, 2018-04-08:
Any resolution on this one?

jpudysz, 2018-04-25:
I was able to resolve this issue on my own.
- refactor build.gradle (clean dependencies)
- use `RNCamera` because aspect ratio issue was resolved
- lower quality and resolution

closed 2018-04-25 by jpudysz