react-native-camera/react-native-camera #1381: Execution failed for task ':react-native-camera:processReleaseResources'.

dorman99, 2018-03-22:
RNCamera

### Steps to reproduce
1.
2.
3.

### Expected behaviour
I cant yarn android , any one can help?

### Actual behaviour
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:processReleaseResources'.
> Error: more than one library with package name 'com.google.android.gms.license'

### Environment
- **Node.js version**: v9.1.0
- **React Native version**:react-native: 0.52.0

[image]

linytsysu, 2018-03-22:
#1370

suraj-epsumlabs, 2018-03-22:
I have same issue
[image]

I think this is just because of the com.google.android.gms latest package 12.0.0

crzycoder, 2019-07-03:
> i had same issue and i fixed it by replacing
> 
> ```
> compile 'com.google.android.gms:play-services-vision:+'
> ```
> 
> with
> 
> ```
> compile 'com.google.android.gms:play-services-vision:11.8.0' 
> ```
> 
> in node_modules/react-native-camera/build.gradle

i dont have build.grade at path "node_modules/react-native-camera/build.gradle"
[image]

shrey91, 2019-07-10:
> i dont have build.grade at path "node_modules/react-native-camera/build.gradle"
> [image]

Its actually under react-native-camera/android/build.gradle

But the newer versions dont have the compile 'com.google.android.gms:play-services-vision:+', they use the version 17.0.2 as a fallback. 

I tried removing the variable and added the + but it doesn't seem to affect it.

crzycoder, 2019-07-12:
Have you solved the issue or not if not i can solve it

lior2710, 2019-07-31:
please...