react-native-camera/react-native-camera #1383: GoogleMobileVision/GoogleMobileVision.h file not found

NijaahnandhRV, 2018-03-22:
'GoogleMobileVision/GoogleMobileVision.h' file not found, I have installed the pod file and added the frameworks as given in the instruction even though i get the error
[image]
[image]
[image]

My header search path

[image]

Can you please help me!!!!

jgfidelis, 2018-03-22:
The header search path is wrong. Change the first line to `$(SRCROOT)/../../../ios`

jgfidelis, 2018-03-22:
Or try master branch, I fixed it there.

closed 2018-03-22 by sibelius

danna88, 2018-12-04:
Change the first line to $(SRCROOT)/../../../ios?    is project targets or RNCamera targets?