react-native-camera/react-native-camera #1384: Error: more than one library with package name 'com.google.android.gms.license'

sandropoluan, 2018-03-22:
Everything was fine. Until I recompile the project and got this error. 

* What went wrong:
Execution failed for task ':react-native-camera:processReleaseResources'.
> Error: more than one library with package name 'com.google.android.gms.license'

stepanvanzuriak, 2018-03-22:
Same for me, and even this not work 
```
configurations.all {
    resolutionStrategy {
       // Add force (11.4.0 is version you want to use)
       force "com.google.android.gms:play-services-gcm:11.8.0"
    }
}
```
`> Error: more than one library with package name 'com.google.android.gms.license'`

chenxiong666, 2018-03-22:
I have change two place  
1.node_module /react-native-camera/android/build.gradle 
    compile 'com.google.android.gms:play-services-vision:11.8.0'

2.  app/build.gradle 
       compile (project(':react-native-camera')) {
    	exclude group: "com.google.android.gms"
	compile 'com.android.support:exifinterface:27.+'
	}
    compile ("com.google.android.gms:play-services-vision:11.8.0") {
    	force = true;
	}
I don't think two versions of the same version will be wrong

anandwahed, 2018-03-22:
This PR #1380 should fix the issue and you need to update your project `android/build.gradle` file to pass the required version of the library.

ThaJay, 2018-03-22:
https://stackoverflow.com/a/49418244/5243543

```def googlePlayServicesVersion = '11.8.0'

allprojects {
    configurations.all {
        resolutionStrategy {
            eachDependency { DependencyResolveDetails details ->      
                if (requested.group == 'com.google.android.gms') {
                    details.useVersion "$googlePlayServicesVersion"
                }
                if (requested.group == 'com.google.firebase') {
                    details.useVersion "$googlePlayServicesVersion"
                }
            }
        }
    }
}

```

dauane, 2018-03-22:
Thank you @ThaJay . It worked perfectly.

closed 2018-03-22 by sibelius

michaelmika, 2018-03-27:
After @ThaJay Solution I got a Dex-Jumbo Error which could be resolved by [https://stackoverflow.com/questions/17320088/android-build-dex-jumbo-mode-in-gradle](this)