react-native-camera/react-native-camera #1387: Build Failed with exception Error: More than one library with package name 'com.google.android.gms.license'

membiblio, 2018-03-22:
### Which implementation are you using
react-native-camera 1.0.2

### Steps to reproduce
1. react-native init MyCamera 
2. npm --save install react-native-camera
3. react-native link
4. Add maven lines to android/build.gradle including jitpack and maven.google.com

### Does it work with Expo Camera?
RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
You should open an issue there as well, so we can cooperate in a solution.

I don't use expo so I am unsure if it works with expo

### Expected behaviour
Tell us what should happen

The react-native run-android command should complete without errors and launch the app 

### Actual behaviour
Tell us what happens instead

I receive the error:
----------------------------------------------------------------------------------------------------------------
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:processReleaseResources'.
> Error: more than one library with package name 'com.google.android.gms.license'
----------------------------------------------------------------------------------------------------------------

### Environment
- **Node.js version**:  v6.13.1
- **React Native version**:  react-native: 0.49.3   (tried with latest as well with same results)
- **React Native platform + platform version**:  Android 6.0.1 is test device and target

### react-native-camera
**Version**: npm version or "master"
1.0.2

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

If I can get this app working I surely will :)

maxschmeling, 2018-03-22:
This error just started happening for me too.

maxschmeling, 2018-03-22:
Duplicate of #1384

closed 2018-03-22 by sibelius