react-native-camera/react-native-camera #1389: Front camera recorded video is still upside down

luisrobertosilvaguillen, 2018-03-22:
Hi, i have an issue trying record a video, the video resulting is upside down, i upgrade the cameraview repo building with d529251d24c0a367a28eea28f0eac2269d12f772 version, but the bug is not fixing in my case.

### Which implementation are you using
RNCamera

### Expected behaviour
Record a video with the front camera of the phone and then play it.

### Actual behaviour
The video is recorded but when I played it is upside down although the preview was the right way.

### Environment
- **Node.js version**: v8.8.0
- **React Native version**: 0.54.2
- **React Native platform + platform version**:  Android 7.0, Android 6.0

### react-native-camera
**Version**: ^1.0.2"

sibelius, 2018-03-22:
use master

chad187, 2018-07-20:
I'm using 1.1.5 the current master. I checked to make sure that the other fixes mentioned are reflected in my code and they are. I still have videos that I take with the front camera portrait turn upside down with playback. I have moto x4 on android 8.1.0

sean-elder123, 2018-09-13:
Any further updates on this?

closed 2019-06-08 by stale[bot]