react-native-camera/react-native-camera #1390: Shutter shoud fired twice

sandropoluan, 2018-03-23:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using
RCTCamera

### Expected behaviour
Shutter only fired one time

### Actual behaviour
Shutter fired twice every time I run `capture()` method

### Environment
- **React Native version**: 0.53.3
-Android all vesion

### react-native-camera
**Version**: 1.0.1
```
takePhoto=()=>{
    const data = await this.camera.capture().then(data=>{
      console.log(data)
    }).catch(error=>{
      toast(error);
    });
}

render(){
    return(
    <View style={{flex:1}}>
        <Camera
        ref={(cam) => {
        this.camera = cam;
        }}
        style={{flex: 1,justifyContent: 'center',alignItems: 'center',flexDirection:'column'}}
        aspect={Camera.constants.Aspect.fill}                                                                                                                                
        type={this.state.type}
        >
            <TouchableOpacity onPress={this.takePhoto} style={{width:70,height:70,borderRadius:35,backgroundColor:'#1ab394',justifyContent:'center',alignItems:'center'}}>
                <MaterialIcons name='camera' style={{color:'#FFF',fontSize:40}}/>
                </TouchableOpacity>
        </Camera>
    </View>);
    }
```

This happen 80% of the cases.  

I don't use RNCamera due to [this](https://github.com/react-native-community/react-native-camera/issues/1267) problem

jpudysz, 2018-03-23:
I encountered same issue on GoClever Quantum 2

sibelius, 2018-03-28:
is this happening on RNCamera?

closed 2019-04-18 by sibelius