react-native-camera/react-native-camera #1399: How to enable the camera face detector feature

harris666, 2018-03-26:
version:
"react": "16.2.0"
"react-native": "0.51.0"
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git"

platform: android

I've written 'onFacesDetected' on RNCamera , but still not calling the callback.  What should i do?

jgfidelis, 2018-03-27:
Check our example app
https://github.com/react-native-community/rncamera-example

closed 2018-03-27 by jgfidelis

harris666, 2018-03-28:
The example still not work.

piercefreeman, 2018-06-04:
Anyone have additional information about this?

FrikkieSnyman, 2018-06-05:
I am experiencing the same problem.

On my Google Pixel 2, the `onFacesDetected` callback works perfectly. But executing the exact same code on a Samsung Galaxy Tab A6, the callback is just not being called at all. What is strange, is that the faceDetectionError callback is also not being called.

Would love to get a solution to this?

stewartduffy, 2018-08-17:
Any updates on this??

webmasterdragon, 2018-09-15:
Same issue here.

drhenault, 2018-10-02:
Same here. Works with a Samsung Galaxy S6, but not with a Pixel C tablet.

idrissbellil, 2018-10-11:
are testing on Android 8.0.0 and up?

ManigandanRaamanathan, 2019-01-18:
Any update on this issue?

liguanhan, 2020-04-27:
OnFacesDetected no callback. Are there any updates?