react-native-camera/react-native-camera #1400: Camera Not Found after navigating back to the camera component

dorman99, 2018-03-26:
so.. i made a case where my user login using google vision, if the data didnt match up with the requirement , they well throw back to then first landing page, and have to use the camera again

the problem is, the camera was undetected if the case above was happend, possbile undhanled prommise rejection : no camera found. that was the error

any suggest , did i have to unmount the camera after using it and how...

pls help !!

jgfidelis, 2018-03-28:
Are you popping back the stack or are you navigating to the screen again?

My guess is (it happened in a project of ours): you are pushing a new screen to the stack and on Android you cannot have two cameras mounted at the same time. The second one does not mount, exactly your scenario. You have to pop back or reset your stack.

Bartlebyy, 2018-10-08:
I had a similar problem. The camera component would not unmount when I switched between tabs. So when I returned back to the cameraScreen it would just be black. So I took advantage of the `componentDidMount` and added two listeners `willFocus` and `willBlur` and now it works. 
``` 
 componentDidMount() {
    const { navigation } = this.props;
    navigation.addListener('willFocus', () =>
      this.setState({ focusedScreen: true })
    );
    navigation.addListener('willBlur', () =>
      this.setState({ focusedScreen: false })
    );
  }

  render() {
    const { hasCameraPermission, focusedScreen } = this.state;
    if (hasCameraPermission === null) {
      return <View />;
    } else if (hasCameraPermission === false) {
      return <Text>No access to camera</Text>;
    } else if (focusedScreen){
      return (this.cameraView());
    } else {
      return <View />;
    }
  }
```

sibelius, 2018-10-08:
@Bartlebyy would you mind send this as a new .md inside docs/

I think this could help a lot newcomers

enisinanaj, 2019-04-15:
Pretty neat. I'm using `react-native-navigation` anyway the same approach works for me as well!

closed 2019-04-29 by dorman99

akhan118, 2019-07-24:
it might it helpful to someone else, this is how I went about it using react-native-navigation.

```
  componentDidAppear() {
    this.setState({ focusedScreen: true })
  }

  componentDidDisappear() {
    this.setState({ focusedScreen: false })
  }
```

```

  render() {
    const { hasCameraPermission, focusedScreen } = this.state;

    if (focusedScreen){
      return <View style={styles.container}>{this.renderCamera()}</View>;
        } else {
          return <View />;
        }
  }
```