react-native-camera/react-native-camera #1404: Android bar code read not workig and camera preview streched

ShahoG, 2018-03-27:
### Which implementation are you using
*RNCamera*

### Steps to reproduce
1. Open Camera
2. Scan bar code

### Expected behaviour
Bar code should be read and data sent to function and camera should have a correct preview image

### Actual behaviour
Nothing happens on bar code scan and camera preview image is streched

### Environment
- **Node.js version**: 8.9.3
- **React Native version**: 0.52.0
- **React Native platform + platform version**: iOS 10.3.1, Android 7.0

### react-native-camera
**Version**: master

I'm testing the same code on iOS and it's working fine, but not on Android (Samsung galaxy edge S7 or any other device). I have it set to type EAN-39 code in the example, but QR o EAN-13 aren't working either.

My code in the render:
```
<View style={{ flex: 1 }}>
   <RNCamera
      ref={ref => { this.camera = ref }}
      onBarCodeRead={data => this._scanOrder(data)}
      style={styles.cameraView}
      barCodeTypes={[RNCamera.Constants.BarCodeType.code39]}
     />
</View>

ShahoG, 2018-03-27:
Also the QR read is extremely slow on android as in iOS it is extremely fast

Swappdevelopment, 2018-03-27:
Having the same issue here. Working perfectly on iOS but on Android it's not. 

React Native version: 0.54.2

ShahoG, 2018-03-28:
Already tried that, getting the same result

brunoksato, 2018-03-28:
I have the same problem. I'm starting debugging android now if I find the problem I send a PR.

Swappdevelopment, 2018-03-28:
I've been able to solve my issue. The problem I had was that my <RNCamera> component was in an absolute view having a width and height of 100%. I've changed the 100% width and 100% height to the device width and height! Try this by just adding a fix width and height!

brunolemos, 2018-03-28:
Stretch issue is duplicated of this: #1243
Please focus on the other issue here

brunolemos, 2018-03-28:
Stretch issue is duplicated of this: #1343
Please focus on the other issue here

closed 2018-03-28 by sibelius

jgfidelis, 2018-04-02:
Please try on master branch

ShahoG, 2018-04-03:
Yes can confirm that this is now working, thanks a lot! However the barcode scanning is a lot faster and more accurate on iOS.

ShahoG, 2018-04-03:
I can also add that QR code scanning is only slow on larger QR codes. Ones containing a small amount of info are relatively fast on android too.