react-native-camera/react-native-camera #1407: Argument list too long: recursive header expansion failed

nikolas7892, 2018-03-28:
### Which implementation are you using
I'm using RNCamera

### Steps to reproduce
1. following documentation ( npm install react-native-camera --save)
2. react-native link
3. Manual link...

### Expected behaviour
I get a build fail and see the following

Argument list too long: recursive header expansion failed at /Users/*USER*/*APP*/node_modules/react-native-camera/ios/../../../ios/build/Index/DataStore/v5/records/1G.

### Actual behaviour
***FAILED BUILD***

### Environment
- **Node.js version**: LATEST
- **React Native version**: LATEST
- **React Native platform + platform version**: iOS 9.0

### react-native-camera
**Version**:  "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jim-at-jibba, 2018-03-28:
I am getting the same issue.

Fingers crossed for a quick fix.

Thanks for all your stellar work on this project.

jim-at-jibba, 2018-03-28:
@nikolas7892 I was able to get my project to build by removing the build folder (`./ios/build`). Hope that is some help.

jgfidelis, 2018-03-28:
@jim-at-jibba did this solve permanently your problem?

jgfidelis, 2018-03-28:
Try to clean the project, restart xcode and build.

jim-at-jibba, 2018-03-29:
@jgfidelis I solved my problem enough to push a build to Testflight. That was the end of the day so we will see this morning 😄

MaheshNandam, 2018-04-05:
I have added Pods at frame relative paths and also [replace master repo at package.json](https://github.com/react-native-community/react-native-camera#how-to-use-master-branch)

but still:
Argument list too long: recursive header expansion failed at ../node_modules/react-native-camera/ios/../../../ios/Pods/React/node_modules/fsevents/node_modules/path-is-absolute.

MaheshNandam, 2018-04-06:
I solved by replace "$(SRCROOT)/../../../ios" at both framework search and header search paths with "$(SRCROOT)/../../../ios/Pods/react-native-camera"

and added "React and Yoga" to my pod file, works for me.

  pod 'React', :path => '../node_modules/react-native', :subspecs => ['Core']
  pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'

mypark, 2018-04-24:
that didn't do the fix for me - anyone have alternative suggestions on a fix using a manual install instructions?

MaheshNandam, 2018-04-25:
@mypark,

Remove/Uninstall 'react-native-camera' from  podfile,

and if Pod target file has 'react-native-camera', remove this as well and try.

neogeno, 2018-04-26:
Deleting the DerivedData folder manually worked for me

M1chaelChen, 2018-04-29:
Tried deleting build folder and rebuild, also tried deleting DerivedData. Still got this error when building.
Anyone got other solutions to this?

cihadturhan, 2018-05-05:
Same here, it was working fine before I integrated firebase.
Is there anyone using react-native-firebase with react-native-camera here?

JJMoon, 2018-05-08:
I got the same error, after install Carthage module for the first time.
Header expansion failed on Carthage module. 

`
Argument list too long: recursive header expansion failed at /Volumes/Data/Repo/mymodule/node_modules/react-native-camera/ios/../../../ios/Carthage/Checkouts/channel-plugin-ios/Carthage/Checkouts/ReSwift/BuildPhases.
`

romk1n, 2018-05-21:
have the same issue, deleting derived directory doesn't help

ccwukong, 2018-06-12:
I got it fixed after adding **$(SRCROOT)/../node_modules/react-native/React** in the Header Search Paths in RNCamera project.

vgeorge, 2018-06-15:
I had version 0.12.0, then I tried to upgrade to 1.1.4  and got the "Argument list..." error. Tried all solutions in this issue, but only suceeded when I removed all references to RNCamera and RCTCamera in `ios/<project_name/project.pbxproj`, cleaned up the project and installed using CocoaPods. The related commit is here:

https://github.com/aetrapp/mobile/commit/517716c53ca1dc3a621d038fe94cc013e435a9ff#diff-3ec47061a629b55cd10c7e19ced7f003. 

Manual install didn't worked for me, even when I removed `Face Dectector` reference.

divsbhalala, 2018-08-02:
I had version 0.12.0, then I tried to upgrade to 1.1.4 and got the "Argument list..." error.
i just try ALL above solution but not able to fix it

sturmenta, 2018-08-04:
I deleted the "ios/build" folder and executed "react-native run-ios" and it worked!

Irrelon, 2018-08-08:
This happened to me after following the cocoapods based install instructions.

Fixed it by: 

1) Removing all mention of "react-native-camera" from the ./ios/Podfile
2) Manually adding the RNCamera project to my main project's Libraries folder (in xcode right click on Libraries in the project tree left side, select "Add files to <project name>..." then navigate to your project's node_modules/react-native-camera/ios/ and select the RNCamera project
3) Go to your project at the top of the project tree and select it, then go to build phases tab and expand the Link Binary With Libraries section, then click the plus and type "cam" the RNCamera.a will show on the list, add that.

Clean project and rebuild.

godoakbrutal, 2018-08-29:
But I've got this error. I really don't know what is happening now, Can anyone help ?

`Argument list too long: recursive header expansion failed at /Users/FOLDER/APP/node_modules/react-native-camera/ios/../../../ios/Pods/boost-for-react-native/boost/numeric/ublas/experimental.`

rn-version: 0.55.4
rn-cli-version: 2.0.1
react-native-camera: 1.1.4 -> installation by use `react-native link react-native-camera`

onmyway133, 2018-08-29:
I had this issue as well, here is how I fix it

- git reset --hard HEAD~5 // go back n commits and start from scratch
- npm install react-native-camera // run this at root of project
- react-native link react-native-camera // run this at root of project
- pod install // run this inside ios folder

Irrelon, 2018-08-29:
Be very careful when running **git reset --hard**, this is not the command you are looking for if you are in your own repo and don't want to potentially loose any changes. Branch first so you are protected!

dmitry-salnikov, 2018-09-06:
As mentioned above, the issue is caused by recursive header expansion, and could be fixed by narrowing down the search paths scope in `FRAMEWORK_SEARCH_PATHS` and `HEADER_SEARCH_PATHS` Xcode project build settings. These values works for me:

`FRAMEWORK_SEARCH_PATHS = $(SRCROOT)/../../../ios/Pods/Headers/**`
`HEADER_SEARCH_PATHS = $(SRCROOT)/../../react-native/React/**`

The next bash script makes the mentioned fix, taking the path to `RNCamera.xcodeproj/project.pbxproj` as first argument:

```bash
#!/bin/bash
#
# fix-rncamera-search-paths.sh
#
# Fix Frameworks/Headers Search Path build settings in react-native-camera/ios Xcode project
# First argument is path to RNCamera xcode project file, e.g.: 
#   ./node_modules/react-native-camera/ios/RNCamera.xcodeproj/project.pbxproj
#

SRC_FILE="$1"

if [ ! -f ${SRC_FILE} ]; then
    echo "Error: RNCamera xcodeproj not found at path: ${SRC_FILE}"
    echo "Skipping fix"
    exit 0
fi

echo "Fixing RNCamera xcodeproj at path: ${SRC_FILE}"
echo "Check before fix:"

cat "${SRC_FILE}" | grep -A 4 FRAMEWORK_SEARCH_PATHS
cat "${SRC_FILE}" | grep -A 3 HEADER_SEARCH_PATHS

perl -i -p0e 's/FRAMEWORK_SEARCH_PATHS = (.*?);/FRAMEWORK_SEARCH_PATHS = "\$(SRCROOT)\/..\/..\/..\/ios\/Pods\/Headers\/**\/**";/gs' ${SRC_FILE}

perl -i -p0e 's/HEADER_SEARCH_PATHS = (.*?);/HEADER_SEARCH_PATHS = "\$(SRCROOT)\/..\/..\/react-native\/React\/**";/gs' ${SRC_FILE}

echo "Check after fix:"
cat "${SRC_FILE}" | grep FRAMEWORK_SEARCH_PATHS
cat "${SRC_FILE}" | grep HEADER_SEARCH_PATHS

echo "Done"
```

To invoke it after every `npm install`, add `fix-rncamera-search-paths.sh` script to root folder of project, and specify postinstall script in `package.json`:

```js
{
  "scripts": {
    "postinstall": "./fix-rncamera-search-paths.sh ./node_modules/react-native-camera/ios/RNCamera.xcodeproj/project.pbxproj"
  }
}
```

sommesh, 2018-11-16:
@cihadturhan Please let me know have you fixed this issue ..

I'm facing the same issue when integrating firebase ..

jayb967, 2018-12-31:
I am also having issue, have firebase installed as well!

prakhar-goel, 2019-01-16:
Try `react-native link`

n1ru4l, 2019-01-17:
@sibelius Can this by fixed by adjusting the `RNCamera.xcodeproj` or is there a reason why the search paths are recursive?

sibelius, 2019-01-17:
Maybe we can fix here, but it should work on both normal projects and podfile projects

isaaclem, 2019-01-28:
I tried removed `./ios/Build` folder and running command `react-native run-ios`, it's working fine, build successfully and running on simulator.

Now I switched over to XCODE and build it, I'm hitting the same problem again. Anyone have idea how to deal with it?

zq513705971, 2019-03-17:
I remove the **framework search paths** values and add **header search paths** with the path **$(SRCROOT)/../../react-native/React** & **$(SRCROOT)/../../React** in the **RNCamera.xcodeproj** to fix it

Mohammad-Khalid23, 2019-03-25:
**This worked for me.**
You have to edit your Framework Search Paths of RNCamera.xcodeproj. Go to,

Xcode -> Your Project -> Libraries -> RNCamera.xcodeproj -> Build Settings -> Framework Search Paths
Double click to edit and update value from `$(SRCROOT)/../../../ios` to `$(SRCROOT)/../../ios`
If there is both **$(SRCROOT)/../../../ios** and **$(SRCROOT)/../../ios** present then remove the first one.

krewllobster, 2019-03-29:
@Mohammad-Khalid23 Thank you for the detailed instructions! This worked for me after manual ios installation (no pods). For anyone browsing this issue for a solution, try Mohammad's solution first as it is dead simple and well explained.

johanarnor, 2019-04-30:
Had a similar issue which was solved by opening (and building) the `.xcworkspace` file in Xcode (instead of `.xcodeproj`).

hansol775, 2019-05-22:
@Mohammad-Khalid23 Thank you. this solution is the answer. Don't reset the git often.

fouad1994, 2019-06-26:
> I solved by replacing "$(SRCROOT)/../../../ios" at both framework search and header search paths with "$(SRCROOT)/../../../ios/Pods/react-native-camera"
> 
> and added "React and Yoga" to my pod file, works for me.
> 
> pod 'React', :path => '../node_modules/react-native', :subspecs => ['Core']
> pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'

Where did you do these changes please?

sibelius, 2019-08-01:
use autolinking with v3

closed 2019-08-01 by sibelius

gongmingqm10, 2019-09-20:
> **This worked for me.**
> You have to edit your Framework Search Paths of RNCamera.xcodeproj. Go to,
>
> Xcode -> Your Project -> Libraries -> RNCamera.xcodeproj -> Build Settings -> Framework Search [...]

Works for me. Saved my day. Thanks.

jamonholmgren, 2019-10-12:
Just noting that there's a mention of this issue in the official documentation:

https://facebook.github.io/react-native/docs/0.60/troubleshooting#argument-list-too-long-recursive-header-expansion-failed

moxspoy, 2019-10-16:
> **This worked for me.**
> You have to edit your Framework Search Paths of RNCamera.xcodeproj. Go to,
>
> Xcode -> Your Project -> Libraries -> RNCamera.xcodeproj -> Build Settings -> Framework Search [...]

in my case, this way works for me. thank you