react-native-camera/react-native-camera #1408: recordAsync: Expected a Camera component

sivaranjansahu, 2018-03-28:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using
RNCamera

### Steps to reproduce

Trying to record video with recordAsync, using the below code : 

1. <RNCamera
            ref	  = { (cam) => { this.camera = cam; } }
    				style = { styles.preview }
    				type  = { RNCamera.Constants.Type.back }
    				onCameraReady  = { () => { 
    				   this.setState( { message : 'Camera Ready !'})
    				}} />
2.   startVideo = async () => {
    if ( !this.camera ) return;
    try {
		  const data = await this.camera.recordAsync({ });
			console.log( data);
			this.setState( { message: 'Good' });
		} catch ( error) {
		  this.setState( { message: 'bad' });
      throw error;
		}
  }
   
3.stopVideo = () => {
		if ( !this.camera ) return;
		this.camera.stopRecording();
  }
4. react-native run-android
### Does it work with Expo Camera?
Did not try

### Expected behaviour
Video should be recorded and URI should be shown on console

### Actual behaviour
Error in console, app crashes

Console error ::
Possible Unhandled Promise Rejection (id: 0):
Error: recordAsync: Expected a Camera component
Error: recordAsync: Expected a Camera component
    at createErrorFromErrorData (blob:http://localhost:8081/ea79261d-48ec-42ef-b126-fbc7cc730629:2023:17)
    at blob:http://localhost:8081/ea79261d-48ec-42ef-b126-fbc7cc730629:1975:27
    at MessageQueue.__invokeCallback (blob:http://localhost:8081/ea79261d-48ec-42ef-b126-fbc7cc730629:2417:18)
    at blob:http://localhost:8081/ea79261d-48ec-42ef-b126-fbc7cc730629:2162:18
    at MessageQueue.__guardSafe (blob:http://localhost:8081/ea79261d-48ec-42ef-b126-fbc7cc730629:2330:11)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (blob:http://localhost:8081/ea79261d-48ec-42ef-b126-fbc7cc730629:2161:14)
    at http://localhost:8081/debugger-ui/debuggerWorker.js:72:58
console.warn @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\ReactNative\YellowBox.js:82
onUnhandled @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\Promise.js:44
onUnhandled @ D:\Learn\ReactNative\camera2\node_modules\promise\setimmediate\rejection-tracking.js:71
(anonymous) @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\Core\Timers\JSTimers.js:258
_callTimer @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\Core\Timers\JSTimers.js:154
callTimers @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\Core\Timers\JSTimers.js:407
__callFunction @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:353
(anonymous) @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:118
__guardSafe @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:316
callFunctionReturnFlushedQueue @ D:\Learn\ReactNative\camera2\node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:117
(anonymous) @ debuggerWorker.js:72

### Environment
- **Node.js version**: 6.11.4
- **React Native version**: 0.54.3
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
Android  7.1.1
### react-native-camera
**Version**: 1.0.3

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jing-wu, 2018-03-29:
I got this error too. But I solved this by adding the following permissions in AndroidManifest.xml

`
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
`

sivaranjansahu, 2018-03-29:
@jing-wu  I have that already in my manifest. 

I am stuck in this problem for last 2 days and its breaking my head! I am sure there is some small thing which I am missing, may be a combination of builds or something. Can any one please help!

 Here is the file : 

<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.camera2">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
      <uses-permission android:name="android.permission.CAMERA" />

<uses-permission android:name="android.permission.RECORD_AUDIO"/>
<uses-permission android:name="android.permission.RECORD_VIDEO"/>
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:windowSoftInputMode="adjustResize">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
    </application>

</manifest>

jgfidelis, 2018-03-30:
Your camera is not being mounted probably, is onMountError being called?

internet5, 2018-04-02:
I have the same problem
How to resovle it?

internet5, 2018-04-02:
I found the reason of error
the options before: quality: RNCamera.Constants.VideoQuality["2160p"] 
I changed quality to RNCamera.Constants.VideoQuality["460p"]
It worked!!

alzoubiMohammed, 2018-04-02:
I have the same problem
How to sovle it?

jgfidelis, 2018-04-07:
@alzoubiMohammed as @internet5 said, he changed the VideoQuality

closed 2018-04-07 by jgfidelis

internet5, 2018-04-14:
@jgfidelis @alzoubiMohammed @sivaranjansahu 
I  was wrong, that problem doesn't matter with the VideoQuality, it's record-audio permission that is not opend automatically.
It worked when I open the record-audio permission manually.
i have this permissions:
<uses-permission android:name="android.permission.RECORD_AUDIO"/>
Why i didn't get it opened? 
You sovled it?

stevanus1997, 2018-04-18:
i have same problem, you already fix it?
if u fix it please the way

alzoubiMohammed, 2018-04-19:
@stevanus1997  @internet5  
I found the solution only by adding the  Mute option in recordAsync()
like
const options = { mute:false };
 It will work

stevanus1997, 2018-04-20:
const data = await this.camera.recordAsync({
      quality: RNCamera.Constants.VideoQuality['480p'],
      mute: false
    });

its still not working, always return video will 0 sec duration

sosospicy, 2018-05-24:
@arapocket  use `await` in an `async` function like:
`checkPermission = async () => {
  let hasAudioPermission = await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.RECORD_AUDIO)
}`

async/await are new syntax in ES7. [async function](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/async_function)

stevanus1997, 2018-05-27:
@sosospicy i already have permission for recording audio, but the result form recording always become 0 sec, please any one can help? 3 days more my thesis will be collected

stevanus1997, 2018-05-27:
magic after i leave my code 3 weeks, and now its work

blogdaren, 2018-07-14:
it's indeed that PermissionsAndroid.PERMISSIONS.RECORD_AUDIO not granted by user causing this issue.