react-native-camera/react-native-camera #1411: Caused by java.lang.NullPointerException: Attempt to get length of null array

sibelius, 2018-03-28:
```
Caused by java.lang.NullPointerException: Attempt to get length of null array
       at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:58)
       at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:26)
       at android.os.AsyncTask$2.call(AsyncTask.java:295)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
```

VinceBT, 2018-11-09:
Just got a similar error on master:
```
Fatal Exception: java.lang.NullPointerException: Attempt to get length of null array
       at org.reactnative.camera.events.BarCodeReadEvent.serializeEventData(BarCodeReadEvent.java:74)
       at org.reactnative.camera.events.BarCodeReadEvent.dispatch(BarCodeReadEvent.java:63)
       at com.facebook.react.uimanager.events.EventDispatcher$DispatchEventsRunnable.run(EventDispatcher.java:355)
       at android.os.Handler.handleCallback(Handler.java:789)
       at android.os.Handler.dispatchMessage(Handler.java:98)
       at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
       at android.os.Looper.loop(Looper.java:164)
       at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
       at java.lang.Thread.run(Thread.java:764)
```

n1ru4l, 2018-11-12:
@VinceBT https://github.com/react-native-community/react-native-camera/pull/1916 this might solve your issue.

closed 2019-06-08 by stale[bot]