react-native-camera/react-native-camera #1415: RNCamera 1.0.3 crashes on Android after addition of text recognition.

kraffslol, 2018-03-29:
### Which implementation are you using

RNCamera

### Steps to reproduce
1.  Start out at react-native-camera@1.0.2
2. yarn install react-native-camera@1.0.3
3. Try to use the camera. It will likely crash on android.

### Does it work with Expo Camera?
Yes.

### Expected behaviour
I expect the camera to start up without crashing the app.

### Actual behaviour
Initializing the camera will crash the app on android. Tried to find what caused it by going from commit to commit and found the culprit to be #1388.  Logcat error:

```
03-29 12:15:18.565 27740 27740 E AndroidRuntime: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/android/gms/internal/zzbko;
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.google.android.gms.vision.text.TextRecognizer$Builder.<init>(Unknown Source)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at org.reactnative.camera.RNCameraView.<init>(RNCameraView.java:83)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at org.reactnative.camera.CameraViewManager.createViewInstance(CameraViewManager.java:53)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at org.reactnative.camera.CameraViewManager.createViewInstance(CameraViewManager.java:16)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:44)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:224)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:153)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:813)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:926)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:47)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:986)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:872)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at android.view.Choreographer.doCallbacks(Choreographer.java:686)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at android.view.Choreographer.doFrame(Choreographer.java:618)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:860)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at android.os.Handler.handleCallback(Handler.java:751)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:95)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:154)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:6119)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
03-29 12:15:18.565 27740 27740 E AndroidRuntime: Caused by: java.lang.ClassNotFoundException: com.google.android.gms.internal.zzbko
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at java.lang.VMClassLoader.findLoadedClass(Native Method)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at java.lang.ClassLoader.findLoadedClass(ClassLoader.java:742)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at java.lang.ClassLoader.loadClass(ClassLoader.java:362)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
03-29 12:15:18.565 27740 27740 E AndroidRuntime:        ... 25 more
03-29 12:15:18.565 27740 27740 E AndroidRuntime: Caused by: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/android/gms/common/internal/safeparcel/zza;
```

### Environment
- **Node.js version**: 9.4.0
- **React Native version**: 0.53.3
- **React Native platform + platform version**: Android 7.1.1

### react-native-camera
**Version**: 1.0.3

### Potential solution
I managed to fix this issue by changing my build.gradle file. Seems like the text recognition part isn't in play-services-vision version 10.2.0 which is the default for this package.

Changing from 

```gradle
compile (project(':react-native-camera')) {
    exclude group: "com.google.android.gms"
    compile 'com.android.support:exifinterface:27.+'
    compile ("com.google.android.gms:play-services-vision:10.2.0") {
  	    force = true;
   }
}
```

to 

```gradle
compile (project(':react-native-camera')) {
    exclude group: "com.google.android.gms"
    compile 'com.android.support:exifinterface:27.+'
    compile ("com.google.android.gms:play-services-vision:12.+")
}
```

fixed the issue for me.

sibelius, 2018-03-29:
@kraffslol can u send a PR adding that we need version 12 for text detection?

jgfidelis, 2018-03-30:
can you tell me if #1418 helps you?

kraffslol, 2018-03-30:
Tested it right now. Using 12.0.1 instead of 12.+ in my build.gradle seems to work just fine indeed. 👍

closed 2018-03-30 by jgfidelis

fatfatson, 2018-04-14:
my case:
react-native-device-info use `gms:+` and now it's 15,
so even with 12, react-native-camera won't get its gms loading because there's only one highest version packed into APK.
after changing it to `+` solved this.

max8392, 2018-06-14:
I solved   
is Important : `compile 'com.android.support:exifinterface:**27**.+`
Change lines :
    `compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        compile 'com.android.support:exifinterface:26.+'
        compile ('com.google.android.gms:play-services-vision:12.0.1'){
            force= true
        }
    }`

to 
` compile project(':react-native-device-info')
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        compile 'com.android.support:exifinterface:27.+'
        compile ('com.google.android.gms:play-services-vision:12.+'){
            force= true
        }
    } `

PR #1418 (2018-03-30): chore(gms): change default gms to 12.0.1
try to close #1415

thought about putting `12.+` but I am afraid of putting stuff with + and leading to crashes :D 
Hope changing this version won't be a big deal.

If this is merged and stable, maybe we can release rncamera 1.1.0, since text recognition would be too much for a 1.0.4, in my opinion.