react-native-camera/react-native-camera #1416: Touch to focus for RNCamera

sri, 2018-03-29:
### Which implementation are you using
*RNCamera*

### Steps to reproduce
1. Tap the Camera
2. RCT Camera: focuses on the point touched and draws a rectangle on the screen for a few secs.
When talking pics, when you touch the darker areas, it sets the focus to that region and so the image brightens up. I'm guessing this is because of the exposure stuff here: https://github.com/react-native-community/react-native-camera/blob/9dbfdff53d2604896f9962a4f8276c9bcc255b85/ios/RCT/RCTCameraManager.m#L1081
3. RN Camera: doesn't do anything

### Does it work with Expo Camera?
No.

Was there a reason the touch-to-focus was removed? 
From reading the source, looks like there are no props that I can pass in to mimic that functionality of `RCTCamera`.

And to add it, seems like I have to port over this method:
https://github.com/react-native-community/react-native-camera/blob/master/ios/RCT/RCTCameraManager.m#L1064

Wondering if this is an easy thing to do?
Or am I missing something?

Thanks!

kesha-antonov, 2018-07-14:
+1 need it

pribeh, 2018-07-22:
@sri Have you taken a stab at this yet? Looks like we'll need this as well.

sri, 2018-07-29:
@pribeh I haven't started on it and won't be able to get to it anytime soon.

closed 2018-09-25 by n1ru4l

pedrosimao, 2019-01-07:
Hello, 
Can you leave this issue open as a "Feature Request"?
I could try to make a pull request myself, by using `autoFocus` together with `autoFocusPointOfInterest`.
The only problem is that `autoFocusPointOfInterest` does not seem to exist on Android.
Can you confirm this last information?

Craigtut, 2019-01-07:
As of this point, the master branch does not contain autoFocusPointOfInterest for android.

I opened #1974  a while back adding autoFocusPointOfInterest to android, but as of now it hasn't been reviewed and added. It's a few releases behind master at this point.

fekajin, 2019-11-21:
> As of this point, the master branch does not contain autoFocusPointOfInterest for android.
> 
> I opened #1974 a while back adding autoFocusPointOfInterest to android, but as of now it hasn't been reviewed and added. It's a few releases behind master at this point.

is it on master now?

Craigtut, 2019-11-21:
@fekajin - Yes, Its been included since version 2.2.0

vetrivendhan48, 2020-04-23:
Let me know @n1ru4l , is this workaround will help us

vetrivendhan48, 2020-04-23:
Let me know if this workaround (pseudo code) will help us @n1ru4l 

```                       
 <View style={{ flex: 1 }}
                     onLayout={(e) => ...store 'height' and 'width' in a state .... }
                        onTouchStart={(e) => ...store 'x' and 'y' point in a state ...}
                        >
                            <RNCamera
                        autoFocusPointOfInterest= { x: x/height, y: y/width }
........
/>
</View>
```

I will soon start this workaround for my Barcode scanner component.
Thanks.

Hajan39, 2021-08-21:
> I am planning to implement the following workaround @n1ru4l
> 
> ```
>  <View style={{ flex: 1 }}
>    onLayout={(e) => ...store 'height' and 'width' in a state .... }
>    onTouchStart={(e) => ...store 'x' and 'y' point in a state ...}
> >
>     // values from the state
>     <RNCamera
>         autoFocusPointOfInterest= { x: x/height, y: y/width }
>         ........
>    />
> </View>
> ```
> 
> I will soon start this workaround for my Barcode scanner component.
> Did we have 'TouchToFocus' implemented again?.
> Thanks.

I think it should be  autoFocusPointOfInterest= { x: x/width, y: y/height }

vetrivendhan48, 2021-08-21:
Thanks @Hajan39. You are right, its my mistake. I edited my workaround.

PR #1818 (2018-09-25): Auto focus point of interest
Closes #1640
Closes #1416
Closes #826