react-native-camera/react-native-camera #1420: SHUTTER SOUND EFFECT OR SHUTTER VISUAL EFFECT

madebytilde, 2018-03-30:
We have done a lot of work with RN Animated library in our enhanced version of camera... but it does not look perfect because there is something too small and very essential that is missing… SHUTTER SOUND EFFECT OR SHUTTER VISUAL EFFECT… Without this small detail the whole screen looks like a mediocre… I have researched regarding this issue and I have seen your tendency to avoid such a topic… I do not know why… open the camera in any app that you have installed and take a picture… instagram, facebook, twitter, snapchat... when you click the button to take a picture you have to hear a sound and see an visual effect… In 2018 this is not luxurious, this is essential… 
We have also spoken in the link below regarding this issue… 
https://github.com/react-native-community/react-native-camera/issues/1378

jgfidelis, 2018-03-30:
We would accept a pr with this feature. We are not avoiding anything. This is implemented on RCTCamera, you can base your pr on its implementation. Its just a matter of someone sending a pr, its the beauty of open source...

egold, 2018-03-31:
FWIW there is a shutter sound effect, it's just not documented anywhere.

```
playSoundOnCapture={true}
```

As for the shutter visual effect, just draw an overlay on the screen and have it fade out over some number of milliseconds that works for you. No need to shout at people making free, open source software.

jgfidelis, 2018-03-31:
@egold I believe this prop is available on RCTCamera only, but it would be easy to copy the implementation to rncamera

jgfidelis, 2018-04-06:
implemented by #1441

closed 2018-04-07 by jgfidelis