react-native-camera/react-native-camera #1421: Camera Component when rendered crashes the app/force closes

kanalasumant, 2018-03-30:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?
RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Actual camera should be rendered when the component mounts.

### Actual behaviour
When the following Camera component is rendered, it shows up the component for a split second but the app force closes. Maybe something to do with permissions, since I didn't see any camera permissions setup in the instructions linking to the AndroidManifest.xml file except for audio, read and write.
Below is the code which works on an ipad running IOS v9.3.5. I'd appreciate it if anyone could suggest me if I'm missing something obvious.
```
takePicture = async function() {
  if (this.camera) {
    const options = { quality: 0.5, base64: true };
    const data = await this.camera.takePictureAsync(options);
    console.error(data.uri);
  }
};

export default ({ goToGalleryView }) => (
  <View style={styles.container}>
    <RNCamera
      ref={ref => {
        this.camera = ref;
      }}
      style={styles.preview}
      type={RNCamera.Constants.Type.back}
      flashMode={RNCamera.Constants.FlashMode.on}
      permissionDialogTitle={"Permission to use camera"}
      permissionDialogMessage={
        "We need your permission to use your camera phone"
      }
    />
    <View style={{ flex: 0, flexDirection: "row" }}>
      <View style={{ flex: 1 }}>
        <ActionButton goToView={goToGalleryView} icon_name="times" />
      </View>
      <View style={{ flex: 1, justifyContent: "center" }}>
        <TouchableOpacity
          onPress={takePicture.bind(this)}
          style={styles.capture}
        >
          <Text style={{ fontSize: 14 }}> SNAP </Text>
        </TouchableOpacity>
      </View>
      <View style={{ flex: 1 }} />
    </View>
  </View>
);

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: "column",
    backgroundColor: "black"
  },
  preview: {
    flex: 1,
    justifyContent: "flex-end",
    alignItems: "center"
  },
  capture: {
    flex: 0,
    backgroundColor: "#fff",
    borderRadius: 5,
    padding: 15,
    paddingHorizontal: 20,
    alignSelf: "center",
    margin: 20
  }
});
```
### Environment
- **Node.js version**: 8.9.4
- **React Native version**: 0.52.2
- **React Native platform  platform version**: Android 7.0

### react-native-camera
**Version**: 1.0.3
> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

kraffslol, 2018-04-01:
In your app/build.gradle try changing
```
compile ('com.google.android.gms:play-services-vision:10.2.0')
``` 
to
```
compile ('com.google.android.gms:play-services-vision:12.0.1')
```

kanalasumant, 2018-04-04:
@kraffslol it worked. Not sure what it does but if you don't mind could you shed some light on what exactly is happening here. Does this mean devices with lower version of android will not be supported or is this the maximum supported version that needs to be entered so that all the latest and greatest devices can run this? Thank you :)

kraffslol, 2018-04-07:
react-native-camera 1.0.3 added support for text recognition. Text recognition requires a more recent version of Google Mobile Vision API than what was specified in the readme at the time. The readme has since been updated to reflect this. Lower versions of android should still work as usual.

closed 2018-04-12 by jgfidelis