react-native-camera/react-native-camera #1424: camera is crash when running on the real machine

cuiyueshuai, 2018-04-02:
I successfully integrated the react-native-camera into my app, but It's crash when I test the example in the real machine, the example from [RNCamera Docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md):

```javascript
class CameraController extends Component {
  static navigationOptions = {
    title: '拍照'
  };
  constructor(props) {
    super(props);
    this.takePicture = this.takePicture.bind(this);
    this.state = {};
  }

  takePicture = async function () {
    console.log(this.camera);
    if (this.camera) {
      const options = { quality: 0.5, base64: true };
      const data = await this.camera.takePictureAsync(options);
      console.log(data);

      const param = {
        image: data.base64,
        language_type: 'CHN_ENG',
        detect_direction: false,
        detect_language: false,
        probability: true,
      };
      CameraAction.baiduOcrAiRequest(param)
        .then((response) => {
          Alert.alert(
            '成功',
            JSON.stringify(response)
          );
          console.log(response);
        }).catch((error) => {
          Alert.alert(
            '错误',
            error.message
          );
          console.log(error);
        });
    }
  };

  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.on}
          permissionDialogTitle={'Permission to use camera'}
          permissionDialogMessage={'We need your permission to use your camera phone'}
        />
        <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
          <TouchableOpacity
            onPress={this.takePicture}
            style={styles.capture}
          >
            <Text style={{ fontSize: 14 }}> SNAP </Text>
          </TouchableOpacity>
        </View>
      </View>
    );
  }
}
```

I just added a request, and should not affect the original code.
when application was running on a real machine, I enter the  photo page and click the SNAP button, It crashed,  the iphone or android's mobile will be stuck on the current page, and I think the phone is a bit hot, so I think application may be stuck in an endless loop.

But what's even more weird is that when the application was running on React Native's Debug mode, everything works. the function `takePictureAsync(options)` can successfully return the result and call the request, So I don't know what happened to this react-native-camera,

Can you help me? Please forgive my bad English

### Environment
node: 8.3.0
react native: 0.54.0
iOS: 11.0.1
android: 5.1

react-native-camera: 1.0.2

jgfidelis, 2018-04-02:
Your English is good.

Is there any crash log available?

cuiyueshuai, 2018-04-04:
@jgfidelis thank very much for you help, I retested the application, the following is the log of operations when the `SNAP` button is clicked.

[image]

Complete log: 
[IphoneLog.txt](https://github.com/react-native-community/react-native-camera/files/1874086/IphoneLog.txt)

I'm sorry, to be honest, I don't understand this log at all,  so I don't know if the logs are actually referenced.

closed 2019-06-08 by stale[bot]