react-native-camera/react-native-camera #1426: Front camera doesn't always start on older Android versions

cristiantone, 2018-04-02:
Hi,

I have an issue with the front camera. On Android version 4.4.2, 5.0, 5.1 the front camera starts only the first time or sometimes it doesn't start at all. Are there any solutions?

### Which implementation are you using
RNCamera

### Expected behaviour
The front camera should start and I should see the preview.

### Actual behaviour
The front camera does not start or starts only once.

### Environment
- **Node.js version**: v4.2.6
- **React Native version**: 0.52.0
- **React Native platform + platform version**: Android 4.4.2, Android 5.0, Android 5.1 etc

### react-native-camera
**Version**:  "master"

jgfidelis, 2018-04-02:
On mount error is being called? Do you have any error log to help?

Ghitu, 2018-04-10:
Any news on this? It's really urgent. Thanks.

sibelius, 2018-04-10:
U need a stacktrace from native code to make this easy to debug

Ghitu, 2018-04-13:
Hello

Can you please help us out with some instructions on how to obtain the stacktrace from native code?

Here is what we got, I'm not sure it's what you are looking for:
[image]

Thank you

sibelius, 2018-04-13:
Run adb logcat

Ghitu, 2018-04-13:
I got this from Android Studio logcat. Maybe it helps.

Also the behavior is like this. The camera shows, I press recording button and recording starts. After pressing stop it goes to other screen with the player. In that screen I have a retake button that goes to the previous route. The camera is black and clicking on the start button it returns this error message Camera is not running. I'm testing on samsung s3 device (GT-I9301I - 4.4.2). 

`D/dalvikvm: GC_FOR_ALLOC freed 1640K, 33% free 12532K/18436K, paused 102ms, total 102ms
E/ViewRootImpl: sendUserActionEvent() mView == null
D/SensorManager: unregisterListener ::   
D/dalvikvm: GC_FOR_ALLOC freed 2184K, 33% free 12395K/18436K, paused 59ms, total 61ms
W/unknown:CloseableReference: Finalized without closing: 42a8a520 42a89be0 (type = CloseableStaticBitmap)
V/MediaPlayer-JNI: isPlaying: 0
    reset
V/MediaPlayer: reset
V/MediaPlayer: message received msg=8, ext1=0, ext2=0
V/MediaPlayer: notify(8, 0, 0) callback on disconnected mediaplayer
V/MediaPlayer-JNI: release
V/MediaPlayer: setListener
    disconnect
V/MediaPlayer: destructor
V/MediaPlayer: disconnect
D/SensorManager: unregisterListener ::   
D/ProgressBar: updateDrawableBounds: left = 0
    updateDrawableBounds: top = 0
    updateDrawableBounds: right = 72
    updateDrawableBounds: bottom = 72
D/dalvikvm: GC_FOR_ALLOC freed 2363K, 35% free 12080K/18436K, paused 32ms, total 33ms
V/MediaPlayer-JNI: native_finalize
    release
E/ViewRootImpl: sendUserActionEvent() mView == null
D/ProgressBar: updateDrawableBounds: left = 0
    updateDrawableBounds: top = 0
    updateDrawableBounds: right = 40
    updateDrawableBounds: bottom = 40
W/DynamiteModule: Local module descriptor class for com.google.android.gms.vision.dynamite not found.
W/DynamiteModule: Failed to load module via fast routecom.google.android.gms.dynamite.DynamiteModule$zza: V2 version check failed
    Local module descriptor class for com.google.android.gms.vision.dynamite not found.
I/DynamiteModule: Considering local module com.google.android.gms.vision.dynamite:0 and remote module com.google.android.gms.vision.dynamite:1702
    Selected remote version of com.google.android.gms.vision.dynamite, version >= 1702
I/Vision: Loading library libocr.so
    libocr.so library load status: true
    Reading ocr models from /data/data/com.google.android.gms/app_vision/ocr/data/models
I/SensorManagerA: getReportingMode :: sensor.mType = 1
D/SensorManager: registerListener :: 0, MPU6500 Accelerometer Sensor, 200000, 0,  
D/dalvikvm: GC_FOR_ALLOC freed 1973K, 35% free 12154K/18436K, paused 61ms, total 63ms
D/MediaScannerConnection: Scanning file 7klwibgf7fvntblfd7(7DCRV7e.(ce../6c(+)6,8da6:.,a6c,+/.fe+-/bc5vh,
V/MediaPlayer-JNI: native_setup
V/MediaPlayer: constructor
    setListener
E/MediaPlayer-JNI: QCMediaPlayer mediaplayer NOT present
W/MediaPlayer: setScreenOnWhilePlaying(true) is ineffective without a SurfaceHolder
V/MediaPlayer-JNI: reset
V/MediaPlayer: reset
V/MediaPlayer-JNI: reset
V/MediaPlayer: reset
V/MediaPlayer-JNI: setDataSourceFD: fd 59
V/MediaPlayer: setDataSource(59, 0, 576460752303423487)
V/MediaPlayer: message received msg=8, ext1=0, ext2=0
    notify(8, 0, 0) callback on disconnected mediaplayer
V/MediaPlayer: setVideoSurfaceTexture
V/MediaPlayer: prepareAsync
V/MediaPlayer-JNI: reset
V/MediaPlayer: reset
V/MediaPlayer: message received msg=5, ext1=720, ext2=1280
V/MediaPlayer: notify(5, 720, 1280) callback on disconnected mediaplayer
V/MediaPlayer-JNI: reset
V/MediaPlayer: reset
V/MediaPlayer: message received msg=1, ext1=0, ext2=0
    notify(1, 0, 0) callback on disconnected mediaplayer
    message received msg=8, ext1=0, ext2=0
    notify(8, 0, 0) callback on disconnected mediaplayer
V/MediaPlayer-JNI: setDataSourceFD: fd 59
[... 40 lines cut ...]
E/ViewRootImpl: sendUserActionEvent() mView == null
D/ProgressBar: updateDrawableBounds: left = 0
    updateDrawableBounds: top = 0
    updateDrawableBounds: right = 40
    updateDrawableBounds: bottom = 40
W/DynamiteModule: Local module descriptor class for com.google.android.gms.vision.dynamite not found.
W/DynamiteModule: Failed to load module via fast routecom.google.android.gms.dynamite.DynamiteModule$zza: V2 version check failed
    Local module descriptor class for com.google.android.gms.vision.dynamite not found.
I/DynamiteModule: Considering local module com.google.android.gms.vision.dynamite:0 and remote module com.google.android.gms.vision.dynamite:1702
    Selected remote version of com.google.android.gms.vision.dynamite, version >= 1702
I/Vision: Loading library libocr.so
    libocr.so library load status: true
    Reading ocr models from /data/data/com.google.android.gms/app_vision/ocr/data/models
I/SensorManagerA: getReportingMode :: sensor.mType = 1
D/SensorManager: registerListener :: 0, MPU6500 Accelerometer Sensor, 200000, 0,  
W/CameraBase: An error occurred while connecting to camera: 1
D/dalvikvm: GC_FOR_ALLOC freed 2081K, 34% free 12331K/18436K, paused 72ms, total 72ms
W/unknown:CloseableReference: Finalized without closing: 42ab76c0 42ab7700 (type = CloseableStaticBitmap)
E/ViewRootImpl: sendUserActionEvent() mView == null
E/ViewRootImpl: sendUserActionEvent() mView == null
D/dalvikvm: GC_FOR_ALLOC freed 1690K, 32% free 12687K/18436K, paused 60ms, total 63ms
W/unknown:CloseableReference: Finalized without closing: 42b5a198 42b5a030 (type = CloseableStaticBitmap)
W/unknown:CloseableReference: Finalized without closing: 42e3aba0 42e453c0 (type = CloseableStaticBitmap)
D/dalvikvm: GC_FOR_ALLOC freed 1883K, 31% free 12760K/18436K, paused 61ms, total 61ms
`

sibelius, 2018-04-13:
try master again, maybe this https://github.com/react-native-community/react-native-camera/pull/1462 fixed your issue

your logs didn't help

0xcodezero, 2018-04-16:
@Ghitu I am facing the same issue, however calling reset instead of goBack() gets not luck to have the camera rendered :( 

I also have tried to use this.props.navigation.replace, but it happens with the same pattern, & the camera not loading.

The issue still happens as you describe with Android versions < 6+, for me It happens on two devices, 
T1-A21L - (Android version 4.4.4)
& Android Samsung Galaxy J3 (Android 5.1.1)

Any guidance on how to make sure about killing the previous screen which has the Camera view, before showing it again.

n1ru4l, 2018-09-24:
This issue seems to be solved. If you still experience this problem please reopen this issue.

closed 2018-09-24 by n1ru4l