react-native-camera/react-native-camera #1430: takePictureAsync : result of image like auto zoom

andiwijayaar, 2018-04-03:
### Steps to reproduce
1.  This is my code
render() {
    if(this.state.mount){
      return (
        <View style={styles.container}>
          <RNCamera
            ref={(cam) => {
              this.camera = cam;
            }}
            ratio={this.state.ratio}
            type={RNCamera.Constants.Type.front}
            style={styles.preview}
            aspect={Camera.constants.Aspect.fill}>
            <Text style={styles.titleCamera}>
              Senyum
            </Text>
            <Text style={styles.titleCount}>
              {this.state.textCount}
            </Text>
          </RNCamera>
          <KeepAwake />
        </View>
      );
    }else{
          return (
              <View style={{flex: 1}}><Image style={{width:'100%', height:'100%' }} source={{uri: this.state.filePath}}/></View> 
          );
    }
    
  }
takePicture = async function() {
    
    if (this.camera) {
      var today = new Date();
      var h = today.getHours();
      var m = today.getMinutes();
      var s = today.getSeconds();

      var fileName = this.state.dataName + "_" + this.state.dataGroup+"_"+h+m+s;
      var saveFile = Constant.PATH+"/"+Constant.APP_FOLDER+"/Images/e.jpg"; 
    
      const options = { forceUpOrientation : true, fixOrientation: true, quality: 1, base64: true, mirrorImage: true};
      const data = await this.camera.takePictureAsync(options)
      this.changeStateFinish(data.uri, data.width, data.height);
      
      const resizedImagePath = data.uri.replace("file:/", "");

      setTimeout(() => {
        RNFS.exists(resizedImagePath).then( (result) => { 
          RNFS.moveFile(resizedImagePath, saveFile)
          .then(() => {
              console.log('RESIZED FILE MOVED');
          })
          .catch((err) => {
              console.log(err.message);
              this.setState({error: true, errorMsg: err.message});
          });
        });

        Database.updateData(this.state.dataId, this.errorCallBack, this.successCallBack);

      
      }, 1000); 
    }
  };
`

2.  camera preview
[image]

3. image result
 
[image]

### Expected behaviour
Image will show with real widht and height (no zoom)

### Actual behaviour
Image show like zoom result

### Environment
- **Node.js version**: 4.6.1
- **React Native version**: 0.54.2
- **React Native platform + platform version**: Android 4.4.2

### react-native-camera
**Version**: 1.0.3 RNCamera

MedinaGitHub, 2018-05-11:
the same happens to me

batuypn, 2018-05-22:
same issue on IOS

batuypn, 2018-05-23:
hi, thanks for the PR. 
Before merge, the result resolution was low, but after merge, i tested the issue. It fixed the image resolution when RNCamera component is **fullscreen**. But in my case, i am using RNCamera inside Scrollview, with a settings view like this: (half height + full height)
[image]
With this configuration, after taking picture, image resolution is **960x781**, normally it should be **960x1280** (like when RNCamera is only child inside Scrollview). So, it cuts the height according to component layout props? I think it should not be like that, RNCamera's layout props should not be related with result image size. 
My render func is:
```javascript
render () {
  return (
    <ScrollView
       ref={sv => this._scrollView = sv}
       horizontal={false}
       pagingEnabled={false}
       showsHorizontalScrollIndicator={false}
       showsVerticalScrollIndicator={false}
       style={styles.scrollViewVertical}
       scrollEventThrottle={32}
       bounces={false}
       directionalLockEnabled
       snapToInterval={SETTINGS_HEIGHT}
       snapToAlignment={'start'}
       decelerationRate={'fast'}
    >
       {this.renderSettings()} // ~half screen height, full screen width
       {this.renderCamera()} // full screen height, width
    </ScrollView>
  )
}
```
If i comment _this.renderSettings()_ method, the result resolution is correct, but with this method  the height is wrong.

batuypn, 2018-05-23:
Yes, i did it like you suggest and it works. 
Thank you @almost

closed 2019-06-08 by stale[bot]