react-native-camera/react-native-camera #1431: Android error in RNCamera

nes123, 2018-04-03:
TypeError: expected dynamic type `double', but had type `null' (constructing arguments for RNCameraModule.takePicture at argument index 1)

any ideas what can it be ?

sibelius, 2018-04-03:
master?

nes123, 2018-04-03:
Yes

sibelius, 2018-04-03:
can u send a fix?

jgfidelis, 2018-04-03:
Your reference is wrong. TakePicture sends two arguments to the native code, the options map and the id of the camera view inside the native view map. It is sending this id as null, so something is wrong with your setup.

nes123, 2018-04-03:
Looking at the documents I see that just the options should be sent to TakePicture: 

takePicture = async function() {
    if (this.camera) {
      const options = { quality: 0.5, base64: true };
      const data = await this.camera.takePictureAsync(options)
      console.log(data.uri);
    }
  };

Am I missing something? I forgot to say that most of the time it works well and the error appears just for some of the users.

jgfidelis, 2018-04-03:
How is your <RNCamera> instantiantion in the render?

nes123, 2018-04-04:
<RNCamera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          type={this.state.camera.type}
          flashMode={this.state.camera.flashMode}
          autoFocus={RNCamera.Constants.AutoFocus.on}
        />

jgfidelis, 2018-04-04:
It looks OK. It is something about the devices... Android is weird :/. You would need to know the device and the android version it is running...

sibelius, 2018-04-04:
Please provide the full stacktrace to help on debug

nes123, 2018-05-12:
Fatal Exception: com.facebook.react.bridge.NativeArgumentsParseException: TypeError: expected dynamic type `double', but had type `null' (constructing arguments for RNCameraModule.takePicture at argument index 1)
       at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:368)
       at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
       at com.facebook.react.bridge.queue.NativeRunnable.run(NativeRunnable.java)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
       at android.os.Looper.loop(Looper.java:154)
       at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
       at java.lang.Thread.run(Thread.java:762)

nes123, 2018-05-25:
any ideas how to fix this issue? i see many errors which are not related to a specific device/OS

sraka1, 2019-03-05:
@sibelius @jgfidelis @n1ru4l Any ideas here? We're also seeing this issue sporadically on Fabric with the exact same stacktrace...

d41z, 2019-04-10:
@sraka1 I have the same problem ... could you solve it?.
@sibelius @jgfidelis @n1ru4l Any ideas here again?

sraka1, 2019-05-02:
I did not manage to fix it yet, however I have reproduced it. There is a condition when you call takePictureAsync while the camera is initializing (so the ref is not null anymore, but the camera is not ready yet) when this occurs. Trying to resolve...

closed 2019-07-01 by sibelius

nthegedus, 2019-08-13:
I've reproduced this error following this steps:

1. Open screen that contains `RNCamera`;
2. **Deny** app take pictures/record video;
3. Try take picture/record video;
4. Crash occurs 

Added a state to manage the camera authorization status
```
  constructor (props) {
    super(props)
    this.state = {
      isCameraAuthorized: false,
    }
  }
```

Update status on RNCamera
```
<RNCamera
  ref={ref => {
    this.camera = ref
  }}
  onStatusChange={(status) => {
    this.setState({ isCameraAuthorized: status.cameraStatus === 'READY' })
  }}
/>
```

Validate camera status before taking picture
```
takePicture = async () => {
    if (!this.state.isCameraAuthorized) {
      this.camera.refreshAuthorizationStatus()
    } else if (this.camera) {
      const image = await this.camera.takePictureAsync({ quality: 0.5, base64: true })
      // DO WHATEVER YOU WANT WITH IMAGE
    }
}
```